hlorus/CAD_Sketcher #1: Easy way to change radius

theoryshaw, 2021-09-19:
Love this plugin!

I'm excited about [this conversation](https://community.osarch.org/discussion/731/geometry-sketcher) and the potential integration to BlenderBIM, as well.

...

Would be nice to be able to change the radius, after the fact, and on the fly. Something like this UI...

[image]

hlorus, 2021-09-20:
Hey there, 
Thanks for the mockup! Are you asking for a way to enter the radius while drawing an arc segment? Otherwise i don't really understand what you mean, why wouldn't you just make a diameter constraint?

theoryshaw, 2021-09-20:
Ah, got it!  Newbie user. didn't see that functionality. awesome.

closed 2021-09-20 by theoryshaw