hlorus/CAD_Sketcher #2: Interactive dimensions

theoryshaw, 2021-09-19:
Related to #1 

would be nice to quickly change the dimension on segments as well... 

Perhaps, being able to 'move' the dimension to indicate which segment you'd like to measure off of. 

[image]

[image]

hlorus, 2021-09-20:
Hey, i assume you're looking for the distance constraint. :)

theoryshaw, 2021-09-20:
Maybe. :)  I see that, but not quite sure how it works, yet. 

video: https://www.dropbox.com/s/49qso0ortut1g2d/2021-09-20_09-31-58.mp4?dl=0

hlorus, 2021-09-20:
Unfortunately this isn't yet documented. As of now a distance constraint is always between a point and a point/line/workplane.

If you're not sure how it works it's best to deselect everything, hit the button in the sidebar and follow the instructions in the statusbar.

theoryshaw, 2021-09-20:
nice... got it!  
When adjusting the dimension, is there anyway to constrain one edge, so the distance changes in one direction only? verses 2 directions, as illustrated in the video below.

video: https://www.dropbox.com/s/jjmd2wcicmchu99/2021-09-20_10-15-14.mp4?dl=0

hlorus, 2021-09-20:
When creating a sketch there should be a fixed point at the origin, it's good practice to define the location to the origin. Alternatively you can right click on a point and fix it.

theoryshaw, 2021-09-20:
very cool.. Thanks for taking me through this. 
Will you use this repo for documentation too?  If so, I can help with PR's here and there.

closed 2021-09-20 by theoryshaw

hlorus, 2021-09-20:
> very cool.. Thanks for taking me through this.
> Will you use this repo for documentation too? If so, I can help with PR's here and there.

I plan to use github wiki for the documentation. Maybe we can discuss the topic on discord?

theoryshaw, 2021-09-21:
Sure!  Since you're in the early days, i would suggest using [Element.io](https://app.element.io/) instead of Discord.  It's open source. :)
Also the [OSarch](https://community.osarch.org/) community uses it all the time, [here](https://app.element.io/#/room/#OSArch:matrix.org).  I could see some awesome cross ferritization happening with that group.