hlorus/CAD_Sketcher #3: Version for Blender 2.79.x (UPBGE 0.2.x)

Symbian9, 2021-09-20:
Backport add-on for latest Blender 2.79.x (UPBGE 0.2.x)

- https://download.blender.org/release/Blender2.79/latest
- https://github.com/UPBGE/upbge/releases/tag/v0.2.5b

hlorus, 2021-09-20:
Hey there,
I'm not sure what changes would be required to support older blender versions. While i think it's possible somehow i don't think this should be a priority right now  as the project is still in an experimental state.

Symbian9, 2021-09-20:
> this should be a priority right now

No, I not ask to prioritize this task, guess its long term feature request.

Maybe I myself would do this work (in partial or in full).

moueza, 2022-07-05:
LTS like Blender 2.93 or 3.2 yes for less work

amrsoll, 2022-08-09:
It seems like the latest UPBGE Is Blender 2.8.x compatible, probably Blender LTS versions would be a good target after the 1.0 release, but I think it would require someone with the need for that to be the maintainer of that feature.