hlorus/CAD_Sketcher #4: Solidifying a sketch

theoryshaw, 2021-09-20:
I might be missing something, but is there a way to correct the 'jagged' result of solidify modifier?
https://user-images.githubusercontent.com/507113/134035213-0180d4e9-b9dc-470f-8590-b8a25f85f41b.mp4

hlorus, 2021-09-20:
I'm not too experienced with modifiers and i currently don't know where this is coming from. When converting to mesh it's using the native to_mesh() function from blender on the generated bezier object (the one you get when setting convert type to bezier).

It usually helps to put a decimate modifier set to planar before doing any other operation.

JanFilipec, 2021-09-20:
It would be great if we could get an ngon mesh instead of triangles

hlorus, 2021-09-21:
> It would be great if we could get an ngon mesh instead of triangles

That's what the planar decimate modifier achieves. Maybe it would be a nice improvement to automatically add it for filled shapes.

JanFilipec, 2021-09-21:
> That's what the planar decimate modifier achieves. Maybe it would be a nice improvement to automatically add it for filled shapes.

Yeah I know, but for example the measureit add on works in the edit mode, where you see the pre modifier triangulated original. You can still select all the triangle faces to label the area for example, but it breaks if you modify the sketch too much. On a single ngon face it works reliably.

hlorus, 2021-09-22:
Referencing geometry of a converted sketch is a different topic. Currently when a sketch is re-converted the already existing bezier/mesh elements are deleted and converting happens from scratch. When you reference a mesh vertex that might correspond to an entirely different BGS point after another convert pass.

Supporting that might be nice but probably not trivial to implement. It would be good to know what use-cases would be allowed by this.

hlorus, 2022-03-11:
This should be fixed since f752bc1f756d4d9f854d6b501d282bd9baec996d.

closed 2022-03-11 by hlorus

sanzoghenzo, 2022-05-28:
Hi, I think I still have this problem, even with the latest commit.

The sketch is fairly simple:

[image]

But when I convert it to mesh and use solidify one of the vertex is offset in the wrong direction.

[image]

I can solve this by setting the solidify mode to Complex, but I wonder if it's just a workaround or a normal behavior (I did very little use of blender in the past, so I don't know how things work anymore :sweat_smile: )

[image]

Here's the sample blender file: [SketchTest.zip](https://github.com/hlorus/CAD_Sketcher/files/8791034/SketchTest.zip)

sanzoghenzo, 2022-05-28:
Ok, the issue here is that the sketch creates 3 edges that weren't supposed to be there; after removing them I can use solidify's  simple mode to get it right:

[image]

Obviously, editing the sketch messes things up again.

I'm a python developer, so I can try to help with this; I'll see if I can do some debugging..

hlorus, 2022-05-28:
Hmm quite strange, i don't know where these edges come from. I will take a look at the sample file at some point.

sanzoghenzo, 2022-05-28:
Just did a debug session, the `mesh_from_temporary` function does a good job at removing the diagonal egdes ((2, 4) and (0, 6) in the picture) from`temporary_mesh`, but doesn't remove the three collinear edges ((2, 5) (1, 5) (1, 6)).

I'm a bit lost in the code flow, but I can see that in the `to_bezier` method the containers have the right segments and points, so I presume that the `sketch.target_curve_object.to_mesh()` creates those edges.

[image]