hlorus/CAD_Sketcher #8: How to start?

mukral, 2021-12-18:
Hi, sorry - thank you very much for you efforts.

I can't find the sketcher module. Would you please give a short hint?

Thank you
Markus

hlorus, 2021-12-21:
If you're on windows you can try running blender as administrator and then press "install from pip" in the addon preferences.
As far as i know installing modules isn't currently possible if blender is installed through steam, snap or similar package managers, maybe that's the problem you're facing...

hlorus, 2022-03-11:
There is now an [installation guide](https://hlorus.github.io/geometry_sketcher/installation/) and a [getting started guide](https://hlorus.github.io/geometry_sketcher/getting_started/) that you can follow if you're still having trouble.

closed 2022-03-11 by hlorus