hlorus/CAD_Sketcher #9: py-slvs not available for python 3.10

Sinasta, 2021-12-28:
Thank you for your great addon!
This is less of a bugreport than a note.
I updated to python 3.10 and it seems like its not possible to install py-slvs on python 3.10.
So this addon is not compatible with python 3.10 it seems to me.
Does somebody know a way to install py-slvs on python 3.10?

hlorus, 2021-12-28:
It looks like the package isn't available for python 3.10 however it's not maintained by me. You can find some information here: https://pypi.org/project/py-slvs/ 
It's probably best to contact @realthunder directly, he was really responsive when i reached out to him.

Anyway it isn't yet relevant for this addon as blender still ships with 3.9.

hlorus, 2022-01-27:
The package seems to be available for 3.10 now so i'll close this issue.

closed 2022-01-27 by hlorus