hlorus/CAD_Sketcher #10: have workplane rotate with object rotation

theoryshaw, 2022-02-13:
**Motivation / Existing Problem**
When you create an extrusion, for example, from a sketch, and then rotate it, and then edit the sketch again, it resets the rotation of the object.

**Description**
When you create an extrusion, for example, from a sketch, and then rotate it, and then edit the sketch again, the sketch's workplan would be orthogonal to the rotation of the extrusion. 

**Mockup**
[image]
[image]
after rotation...
[image]
[image]

hlorus, 2022-02-15:
This is quite a fundamental design topic.

Converted objects currently aren't ment to be translated. Whenever you create a part that consists of multiple sketches the part's local space is simply the world space.

The intended workflow for placing a created part somewhere in 3d would be to create a linked duplicate and transform this instead.

While this would probably support most usecases and could even be automated it indeed wouldn't allow for things like in-place editing and referencing surrounding geometry.

hlorus, 2022-02-15:
I've made a quick test where the workplane is based on a root object. 

https://user-images.githubusercontent.com/64740362/154031336-9f3d202c-9ab8-4812-b3d7-e960d2efd226.mp4

Might be a good approach however such root objects should be supported for all 3d entity types.

theoryshaw, 2022-02-15:
very cool. 

Do you think you could rotate multiple objects at a time, like the following--which happens quite often. 

https://user-images.githubusercontent.com/507113/154074383-c077c94e-6fa8-43c3-ad84-455a8f70a7d3.mp4