hlorus/CAD_Sketcher #13: Dynamically set the workplane based off existing geometry

theoryshaw, 2022-02-13:
**Motivation / Existing Problem**
Currently no way to create a non-orthogonal workplane

**Description**
Pick a rotated workplane based off existing geometry

**Alternative Solutions**
When creating a new sketch, or perhaps even when you edit an existing one, you can dynamically set the workplane based off existing geometry

[image]

hlorus, 2022-02-13:
> Cool. using [image] how do you rotate it to specific dimensions in space?  Can you rotate it after the fact, once it's placed?
> 

You can enter exact angles from the redo panel.

theoryshaw, 2022-02-14:
Thanks @hlorus for sharing the following GIF.  I uploaded it here, for Passerbys...
[image]

hlorus, 2023-01-23:
This functionality is already implemented. Having workplanes move with the mesh they're placed on is tracked in #10.

closed 2023-01-23 by hlorus