hlorus/CAD_Sketcher #14: Copying objects, and copying it's profile along with it.

theoryshaw, 2022-02-14:
**Motivation / Existing Problem**
A lot of modeled objects tend to be started by copying some other object and going from there.  Unfortunately when you copy an object with an associated sketch, that sketch does not get copied as well. 

**Alternative Solutions**
Would be nice if when the objects was copied, a copy of the sketch would go along with it. 
I like the idea of `Profile Types`, that is, if object `A` had Profile `X`, and it was copied to object `B`, that it would still have Profile `X` associated with it. 

That is, if you now changed Profile `X`, both object A and B would change. 

If you copied Object `A` to Object `B`, but wanted to have a different profile associated with object `B`, that you could just duplicate the profile type.  

This could start dove tailing into a workflow around IFCprofiles: https://github.com/IfcOpenShell/IfcOpenShell/issues/1641

theoryshaw, 2022-02-14:
...or maybe there's profile "Types", and profile "Instances".

dmatscheko, 2022-08-23:
@theoryshaw, the behavior you mention (duplicated object is updated by the same sketch) can already be archived by using "Object->Duplicate Linked" in Blender.

However, if this behavior is implemented for normal copy too, it must also be possible to break the connection between an object and its sketch.

This is necessary because otherwise it would be impossible to continue editing an object with default Blender features, except with modifiers. All other changes to an object would be erased each time the object is updated from its sketch.