hlorus/CAD_Sketcher #26: Add integration with IFC

Moult, 2022-03-29:
**Motivation / Existing Problem**
IFC is a way of describing parametric geometry in the world of BIM (Building Information Management - basically the schema for digital buildings). Right now there is no nice UX for modifying these parametric profiles in Blender.

**Description**
I'd love to collaborate with the geometry sketcher plugin to integrate with IFC. Would you be interested in a collaboration or redesigning the plugin so it is more agnostic / generic and can be called from other add-ons too?

Moult, 2022-03-29:
Link to https://github.com/IfcOpenShell/IfcOpenShell/issues/1641

hlorus, 2022-03-30:
Hey there,
Such an integration would be awesome! I'd like to help making that happen.

How would such a redesign be, is there something specific you already imagine?

I think you've said that it should be possible to import the addon or parts of it from another addon, right? How would registration of operators and properties happen in such a scenario?

Moult, 2022-04-11:
I'm not quite sure, I haven't yet had the chance to code dive :) IFC supports defining certain types of geometry. One of these is a polycurve consisting of straight lines and arc segments: http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcIndexedPolyCurve.htm

If I am able to process the IFC polycurve into something that geometry_sketcher can understand, is there some function that can be called to render the profile on the screen so the user can edit it? Once edited, can I insert a callback, listener, hook or perhgaps write my own operator that stops rendering the profile on the screen and receives the edited profile from geometry_sketcher so that I can convert it back into an IFC polycurve?

Moult, 2022-04-26:
Would you be available for a short call to discuss and I can demo how things work in IFC and discuss how an integration might work?

theoryshaw, 2022-04-26:
I would love to see this as well!

Could see CAD_Sketcher being the in-world editor for all [IfcProfileDef](http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcProfileDef.htm)'s

If you need some simple examples from [FreeMVD](https://gitlab.com/osarch/FreeMVD_WorkFlow)

---

[IFCRECTANGLEPROFILEDEF](https://gitlab.com/osarch/FreeMVD_WorkFlow/-/blob/master/Specifications_Test_Files/Extrusions_Rectangles/from_Blender/Extrusions_Rectangles_02.ifc)
schema definition [here](http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcRectangleProfileDef.htm)

```
  /*#237=IFCEXTRUDEDAREASOLID(#231,#235,#236,6.56167979002625);*/
    /*#231=IFCRECTANGLEPROFILEDEF(.AREA.,$,$,5.90551160243344,2.62467200889831);*/
    /*#235=IFCAXIS2PLACEMENT3D(#232,#233,#234);*/
      /*#232=IFCCARTESIANPOINT((2.28083978487751,-2.64041986208888,1.00000011013562));*/
      /*#233=IFCDIRECTION((0.,-0.,-1.));*/
      /*#234=IFCDIRECTION((-1.,0.,0.));*/
    /*#236=IFCDIRECTION((0.,0.,-1.));*/
```

[IFCCIRCLEPROFILEDEF](https://gitlab.com/osarch/FreeMVD_WorkFlow/-/blob/master/Specifications_Test_Files/Extrusions_Circles/from_Blender/Extrusions_Circles_02.ifc)
schema definition [here](http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcCircleProfileDef.htm)
```
  /*#208=IFCEXTRUDEDAREASOLID(#202,#206,#207,2.99999996554407);*/
    /*#202=IFCCIRCLEPROFILEDEF(.AREA.,$,#201,0.499189355954731);*/
      /*#201=IFCAXIS2PLACEMENT2D(#199,#200);*/
        /*#199=IFCCARTESIANPOINT((0.,0.));*/
        /*#200=IFCDIRECTION((1.,0.));*/
    /*#206=IFCAXIS2PLACEMENT3D(#203,#204,#205);*/
      /*#203=IFCCARTESIANPOINT((1.60414813541052E-08,-2.2916401934436E-09,-1.00000069679551));*/
      /*#204=IFCDIRECTION((3.7878029957028E-07,2.31496199987191E-09,-1.));*/
      /*#205=IFCDIRECTION((-0.0578654408454895,-0.998324334621429,0.));*/
    /*#207=IFCDIRECTION((-3.91146457445757E-08,1.79253149212855E-07,-0.999999940395355));*/
```

[IFCARBITRARYPROFILEDEFWITHVOIDS](https://gitlab.com/osarch/FreeMVD_WorkFlow/-/blob/master/Specifications_Test_Files/Extrusions_Void_InnerBoundary/from_Blender/Extrusions_Void_InnerBoundary_02.ifc)
schema definition [here](http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcArbitraryProfileDefWithVoids.htm)
```
  /*#194=IFCEXTRUDEDAREASOLID(#188,#192,#193,3.);*/
    /*#188=IFCARBITRARYPROFILEDEFWITHVOIDS(.AREA.,$,#182,(#187));*/
      /*#182=IFCPOLYLINE((#178,#179,#180,#181,#178));*/
        /*#178=IFCCARTESIANPOINT((-0.500000006179484,-0.49999995729116));*/
        /*#179=IFCCARTESIANPOINT((0.500000006179484,-0.49999995729116));*/
        /*#180=IFCCARTESIANPOINT((0.500000006179484,0.500000348397753));*/
        /*#181=IFCCARTESIANPOINT((-0.500000006179484,0.500000348397753));*/
        /*#178=IFCCARTESIANPOINT((-0.500000006179484,-0.49999995729116));*/
      /*#187=IFCPOLYLINE((#183,#184,#185,#186,#183));*/
        /*#183=IFCCARTESIANPOINT((-0.353553393498806,1.95553296507187E-07));*/
        /*#184=IFCCARTESIANPOINT((0.,-0.353553124613023));*/
        /*#185=IFCCARTESIANPOINT((0.353553393498806,1.95553296507187E-07));*/
        /*#186=IFCCARTESIANPOINT((0.,0.353553613496265));*/
        /*#183=IFCCARTESIANPOINT((-0.353553393498806,1.95553296507187E-07));*/
    /*#192=IFCAXIS2PLACEMENT3D(#189,#190,#191);*/
      /*#189=IFCCARTESIANPOINT((1.77188495325604,0.,2.34991546690933));*/
      /*#190=IFCDIRECTION((0.70710676908493,0.,0.70710676908493));*/
      /*#191=IFCDIRECTION((0.,1.,0.));*/
    /*#193=IFCDIRECTION((0.,0.,1.));*/
```

[IFCARBITRARYCLOSEDPROFILEDEF](https://gitlab.com/osarch/FreeMVD_WorkFlow/-/blob/master/Specifications_Test_Files/Extrusions_Arcs/from_Revit/Extrusions_Arcs_02_PASS.ifc)
schema definition [here](http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcArbitraryClosedProfileDef.htm)
```
          /*#365= IFCEXTRUDEDAREASOLID(#359,#364,#20,3.);*/
            /*#359= IFCARBITRARYCLOSEDPROFILEDEF(.AREA.,'Generic Models 1',#352);*/
              /*#352= IFCCOMPOSITECURVE((#324,#331,#338,#344,#351),.F.);*/
                /*#324= IFCCOMPOSITECURVESEGMENT(.CONTINUOUS.,.T.,#322);*/
                  /*#322= IFCPOLYLINE((#318,#320));*/
                    /*#318= IFCCARTESIANPOINT((-0.500000000000002,-0.5));*/
                    /*#320= IFCCARTESIANPOINT((0.499999999999998,-0.5));*/
[... 6 lines cut ...]
                    /*#332= IFCCARTESIANPOINT((0.5,0.499999999999998));*/
                    /*#334= IFCCARTESIANPOINT((-0.,0.500000000000001));*/
                /*#344= IFCCOMPOSITECURVESEGMENT(.CONTINUOUS.,.T.,#341);*/
                  /*#341= IFCTRIMMEDCURVE(#340,(IFCPARAMETERVALUE(360.)),(IFCPARAMETERVALUE(90.)),.T.,.PARAMETER.);*/
                    /*#340= IFCCIRCLE(#339,0.5);*/
                      /*#339= IFCAXIS2PLACEMENT2D(#10,#28);*/
                        /*#10= IFCCARTESIANPOINT((0.,0.));*/
                        /*#28= IFCDIRECTION((0.,1.));*/
                /*#351= IFCCOMPOSITECURVESEGMENT(.CONTINUOUS.,.T.,#349);*/
                  /*#349= IFCPOLYLINE((#345,#347));*/
                    /*#345= IFCCARTESIANPOINT((-0.500000000000001,0.));*/
                    /*#347= IFCCARTESIANPOINT((-0.500000000000002,-0.5));*/
            /*#364= IFCAXIS2PLACEMENT3D(#360,#362,#18);*/
              /*#360= IFCCARTESIANPOINT((0.353553390593275,0.500000000000007,0.358787223388274));*/
              /*#362= IFCDIRECTION((0.707106781186547,0.,0.707106781186548));*/
[... cut ...]
            /*#20= IFCDIRECTION((0.,0.,1.));*/
```

[IFCISHAPEPROFILEDEF](https://gitlab.com/osarch/FreeMVD_WorkFlow/-/blob/master/Random_Tests/Beam_W_section/from_Blender/Beam_W_section_02.ifc)
schema definition [here](http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcIShapeProfileDef.htm)
```
#541=IFCEXTRUDEDAREASOLID(#75,$,#540,2.);
  /*#75=IFCISHAPEPROFILEDEF(.AREA.,'This is the profile name',$,0.100000001490116,0.319999992847443,0.00499999988824129,0.00999999977648258,0.00499999988824129,$,$);*/
```

Moult, 2022-04-26:
Apart from the primitives and parametric profiles, I reckon the biggest advantage would be the ability to author IfcIndexedPolyCurve which is effectively a polyline which supports arc segments.

hlorus, 2022-04-26:
> Would you be available for a short call to discuss and I can demo how things work in IFC and discuss how an integration might work?

Yeah sure, how about tomorrow (Wednesday) afternoon?

Moult, 2022-05-05:
So sorry I missed your message @hlorus - I'm online in general in the OSArch chatroom during the Sydney timezone. @qwiglydee is also helping hack on the BlenderBIM Add-on so :)

Essentially to integrate with IFC:

 1. What function should we call from the BlenderBIM Add-on to programatically display a profile using cad-sketcher?
 2. What is the format cad-sketcher needs us to describe the input profile to generate any profile that we need? We can build an IFC Profile->CAD Sketcher Profile converter.
 3. After the user modifies that profile what function should we call to stop displaying that editable profile on the screen using CAD sketcher?
 4. What is the output format we can extract of the saved profile? We can build a CAD Sketcher Profile -> IFC Profile converter here.

The first profile I'd like to see being able to convert to and from IFC / CAD Sketcher is an IfcIndexedPolyCurve: http://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcIndexedPolyCurve.htm

It's pretty straightforward, essentially a polyline, where a segment may either be a straight line, or a circular arc. So as long as you can translate to and from a list of sequential points, that's all we need. To represent the arc, it is represented by 3 points - the start, end, and point on the arc (let's go for midpoint).

Does this sound achievable? If you can help point us to some functions to call and examples of inputs and outputs, I can help link it up to IFC and @qwiglydee and write the GUI integration and any necessary math to convert arcs and points and so on.

Moult, 2022-09-25:
This is awesome and your tips made it easy to integrate! Apologies for the delay, I originally wanted to sponsor work by @qwiglydee to work on this but alas it didn't happen.

I've now done a couple prototype operators to allow editing of arbitrary polyline profiles (i.e. no arcs, circles, etc) with CAD sketcher: https://github.com/IfcOpenShell/IfcOpenShell/commit/4c7a5e8a1a9887bd7696f0872920fdebe4eb66cd just to see what the workflow might be like.

It's a promising start but I have some nagging thoughts that suggest it may not be appropriate for some tasks in the BIM world. The BIM world relies a lot more on non-parametric "arbitrary" locations. Just "move this dot here" for example. It also relies a lot of "move this here to align with this edge/corner/whatever of this other 3D object", which CAD sketcher at its current functionality doesn't do very well compared to the OOTB Blender functions of pushing vertices and edges around. I think this means that CAD sketcher should be provided as one option (which might suite well for assembly style part modeling), but there should also be another method more catered towards the BIM world (for the majority of usecases of simple profile editing for slabs, gutters, skirting, mullion extrusion profiles, etc).

hlorus, 2022-09-26:
Great to see your efforts, if i get it right you've implemented two operators which allows to temporarily display and edit geometry with the CS tools. The major downside here is that you loose the parametric nature of the addon as all the constraints are removed between the edits.  

Referencing of meshes and other entities is indeed a big topic which is already on the roadmap however this might take some time to arrive.

hlorus, 2022-09-28:
There's now basic support to write out the addon data of a given scene into a file and load it back in, see: https://github.com/hlorus/CAD_Sketcher/commit/7c548e624a725d6bc3760d4648d92d9e1d882371

Here's a testfile to try it:
[save_load.zip](https://github.com/hlorus/CAD_Sketcher/files/9663508/save_load.zip)

Moult, 2022-09-30:
Awesome :) I've now updated the integration to use the to_dict() and update() methods :) Let's give it a little while to see where users think integration should be pushed, but at least the basics are there.

theoryshaw, 2022-10-06:
Excited about this integration.  I think the following functionality would make a CadSketcher/BlenderBIM integration more attractive and useful. 
- [Add grab operator](https://github.com/hlorus/CAD_Sketcher/issues/254)
- [Snapping to outside geometry](https://github.com/hlorus/CAD_Sketcher/issues/12)
- [Copying objects, and copying it's profile along with it.](https://github.com/hlorus/CAD_Sketcher/issues/14)
- [Dynamically set the workplane based off existing geometry](https://github.com/hlorus/CAD_Sketcher/issues/13)
- [Have multiple sketches for one mesh](https://github.com/hlorus/CAD_Sketcher/issues/299)

franklincg, 2026-08-17:
Hi! I’m interested in working on this bounty. Could you confirm the current bounty amount and whether the issue is still available for a new contributor?

hlorus, 2026-08-17:
As far as i know @falken10vdl is working on this, they see: https://github.com/hlorus/CAD_Sketcher/pull/550