hlorus/CAD_Sketcher #29: Really thin outer lines on MBA M1

mervenca, 2022-04-22:
**Description**
It may be a mac only thing, or resolution dependant- retina 13,3-inch (2560 × 1600)- the outer lines for sketching are too thin to use or select.
It's an issue from the start- really hard to activate a plane when creating sketch (should it also be selectable from the "face" of the plane, not only the edge?), hard to select sides etc to add constraints.

**To Reproduce**
Steps to reproduce the behavior:
1. Using the add-on on Macbook air M1

**Expected behavior**
To select the planes and sketche's edges without a hassle.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - OS:  macOS Monterey 12.3.1
 - Blender Version: 12.3.1
 - Addon Version: 0.20.0
[image]

felixchenier, 2022-04-23:
I also have this problem on an Intel Mac on Catalina.

MarkBTomlinson, 2022-04-23:
It is almost impossible to select work planes and entities on a 16" MBP.
The planes flicker as you hover over but hitting the hot spot is really hard.
I will try this on the PC later to see if it really is a HDPI issue.

NoobTaco, 2022-04-25:
I am seeing this on the Mac Studio Ultra as well.

We need an option to change the line width.

atsuyuki, 2022-04-26:
I found the same issue in another addon. This issue seems to be coming from OpenGL on Mac. It looks like a solution is indicated, but it is beyond me. I hope someone stronger can solve this problem.

[Line weight not working (MacOS)@MeasureIt_ARCH](https://github.com/kevancress/MeasureIt_ARCH/issues/6)

gitblair, 2022-04-27:
While it is possible to choose a plane, it takes many attempts to get there.
The hovering that highlights the plane is not consistent. It feels very similar to attempting to choose anything other than the corner blocks. Also similar to the issue where the lines are too thin to hover over and select.

Mac OS 12.2.1
Apple M1
Blender 3.0.0
Addon 0.20.0

hlorus, 2022-04-29:
This is now mostly resolved, see 4515af54884586dc6619915264d8bac01b651fec.
The only missing part is porting the display of dashed lines to the new shader.

felixchenier, 2022-05-03:
It also seems to work on my mac 😃

MarkBTomlinson, 2022-05-03:
Confirm it is work fine on my intel macBook now.

commit 6621fb98 (2022-05-11): Draw dashed lines in screenspace

Use hardware interpolation to calculate dashinng.
Closes #29, #31
closed 2022-05-11 by hlorus