hlorus/CAD_Sketcher #30: [BUG] wont let me get started

bradfoot, 2022-04-22:
**Description**
the add a sketch button shows the planes but you can't selection, when you click on one they disappear

**To Reproduce**
Steps to reproduce the behavior:
1. start blender
2. hit n to bring up the menu
3. and click on sketcher 
4. click on Add Sketch to bring up the planes
5. click on a plane and they go away

**Expected behavior**
it should make a new sketch and let me draw on it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - OS:  macbook air, m1 chip Mac OS 12.2.1
 - Blender Version: 3.1
 - Addon Version: 0.20.0

felixchenier, 2022-04-23:
You have to select the outer line of the plane, but it is very difficult to do since these outer lines are so thin (#29).

bradfoot, 2022-04-24:
thats a pain but i got it. hope somebody is working on that.
thanks

physicsmazz, 2022-04-24:
Even when I click the outer line, which is next to impossible, it still doesn't start a sketch. Seems like it's possibly a Mac issue.

hlorus, 2022-04-26:
This is a duplicate, see #29. Closing

closed 2022-04-26 by hlorus