hlorus/CAD_Sketcher #31: construction line problem (?)

pgrove604, 2022-04-24:
1. draw rectangle, make it 4 meters on one side and 7 on the other.
2. draw a diagonal line top right to bottom left, and make it a construction line
3. click on the midpoint and the line and make the line a construction line.
4. Resize the box to 20mm on one side and 40mm on the other. The construction line won't reach all the way to the bottom left. 
5. Delete the construction line and draw a new line top right corner to bottom left.
6. Make it a construction line. The new line doesn't go all the way to the bottom left. Repeated attempts will produce the same result.
Expected the original construction line to resize with the rectangle. Otherwise, expected the new construction line to maintain length when converted from a regular line.
Using Blender 3.1.0 and latest version of sketcher (downloaded april 23)
[image]

teddywaweru, 2022-04-25:
1. I don't think that the mid line is a construction line, since construction lines are expected to be dotted.
2. Lines don't have  a defined middle point, so not quite sure what you clicked on in step 3.
3. It's odd that the diagonal line is missing the bottom end point, that's an anomaly.
You could try re-doing the constraints & paying attention to the order that you place them, as this might also bring about some errors in the workflow.

hlorus, 2022-04-25:
I wasn't able to reproduce this, could you share the file?

pgrove604, 2022-04-25:
Unfortunately I deleted the original. However, I've re-created it. Here it is... 
In this I created the rectangle and used a diagonal construction line with a midpoint constraint to center it. Then I added dimensions (Since I do 3D printing I work in millimeters and values were entered as "xxmm" rather than decimals. 
Once resized the construction line is not a construction line and does not extend fully through the rectangle. 
Hope this helps.

pgrove604, 2022-04-25:
I realize that construction lines are supposed to be dotted. The original line was in fact a dotted construction line. However, when I applied dimensions the line became solid and truncated. I deleted the original file, unfortunately, but have recreated it. 
In this I drew the rectangle, then a diagonal construction line (which was a construction line), then used a midpoint constraint on that line and the center to center the rectangle. Up to that point, all was well. Then I applied dimensions in millimeters (since I do 3D printing and work in millimeters) as "xxmm" (this may have caused the problem?) 
Once I finished the second dimension the construction line was no longer a construction line and no longer reached fully across the diagonal. 
Hope this helps

hlorus, 2022-04-26:
@pgrove604 
It looks like the file didn't actually upload...

pgrove604, 2022-04-26:
I've attached it to this email. If it doesn't come through, please let me know. Or is there some other way I should be uploading it?

hlorus, 2022-04-26:
Try opening github in the browser and comment there.

norbini, 2022-04-28:
I'm able to reproduce this issue on a MacBook Pro 2021 16" (Built-in Liquid Retina XDR Display - 3456 x 2234).

I think this might be an issue with the scale of the dash pattern of the construction line when zoomed in the viewport?
At least when zoomed in enough to clearly see a 20mm x 40mm rectangle.

I believe the dimension constraints, and the midpoint constraint, aren't strictly necessary to reproduce the effect seem in the original report - it just requires drawing a sufficiently short construction line (e.g. 0.1m) where the construction line is so short we can only see part of the line's dash pattern.
In the original report, the line appears to become solid since we mostly see the "on" part of the dash pattern, and it appears too short because we also see a small part of the "off" portion of the pattern.

Perhaps the construction line's dash pattern should be independent of the viewport zoom (i.e. so that the pattern is always takes the same number of pixels on the screen, irrespective of the viewport zoom level)?

To reproduce:
1. Start a new sketch, using the X-Y plane.
2. Draw a single line
3. Make it a construction line
4. Move the end points of the construction line closer together, zooming the viewpoint as necessary, until the dash pattern isn't obviously repeating any more.

[short_construction_line.blend.zip](https://github.com/hlorus/CAD_Sketcher/files/8586691/short_construction_line.blend.zip)

I think the following blend file shows the originally reported issue.
Once the length constraints are removed, and a corner freely moved so the rectangle becomes sufficiently large (e.g. a diagonal of 0.13m), it should be possible to see the construction line's dash pattern re-emerge:
[construction_line_does_not_appear_dashed.blend.zip](https://github.com/hlorus/CAD_Sketcher/files/8586719/construction_line_does_not_appear_dashed.blend.zip)

Hope this helps!
Hope this helps?

pgrove604, 2022-04-29:
A scaling issue with construction lines seems to be the cause of this, right enough. I created a 400mm by 500mm rectangle, THEN drew a diagonal and made it a construction line. Showed as two dashes only. This is a problem to me since I work with files of no more than 150 square millimeters. Hence all my lines are measured in millimeters and I'd imagine other 3d modelers will have the same problem.

hlorus, 2022-05-11:
Closed with 6621fb98505c1bc3dd1c3ac8947b09b187a62ba4

closed 2022-05-11 by hlorus