hlorus/CAD_Sketcher #35: [BUG] Mesh rasterization seems to fail for complicated multi-part drawing

JosephCatrambone, 2022-04-24:
**Description**
When converting the following drawing (for finger cutouts in a block), the rasterized mesh does not match the shape of the drawing.

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Seems to happen when using arcs and midpoint constraints.  If I put the center of two arcs on a line and constrain the line to a midpoint (or to some distance from a point) then the solution that materializes is not expected.  Don't have a perfect approach to reproducing yet, but here's another example of a mesh that fails to rasterize:

[image]

[image]

**Expected behavior**
I would expect for a rounded arc to be 

**Screenshots**
Above

**System Information:**
 - OS: Linux Mint 20.04 
 - Blender Version: 3.1.0
 - Addon Version: 0.20.0

Blend File is Attached.

[hand_keyboard.blend.zip](https://github.com/hlorus/CAD_Sketcher/files/8548898/hand_keyboard.blend.zip)

teddywaweru, 2022-04-25:
the issue could be the lines that are joining the arc, & the line between the arcs & the rectangles. I may be wrong but the developers may have set up the add-on to recognize everything apart from construction lines as potential geometry. So this affects  
[image]

so in the above, the left doesn't have the joining line between the arc & rectangle, while the second one does. I'd say the base principle is that any **lines** in the sketch will be regarded as **active geometry** when generating the mesh. All distance constraints would be kept as construction lines

hlorus, 2022-04-25:
@teddywaweru is right here, the converter tries to create closed bezier splines to figure out what area to fill, if there's a junction it has to guess which path to use as the main path. There's some more info in the docs, see:

https://hlorus.github.io/CAD_Sketcher/integration/

Anyway there seems to be another issue so leaving this open for now.

JosephCatrambone, 2022-05-01:
Good catch.  I missed marking that as a construction line.  I wonder what a good way to communicate disconnected paths would be.  🤔  Could perhaps mark 'open' bezier curves in a different color.

hlorus, 2023-01-23:
This seems to be working with recent versions when marking the inner lines as construction.

closed 2023-01-23 by hlorus