hlorus/CAD_Sketcher #39: [BUG] - Add Sketch Planes Impossible to Select

esaeo, 2022-04-26:
**Description**
Simply cannot select a sketch plane when adding a sketch.  It appears the hotspot for selecting is so small any click negates the planes.

**To Reproduce**
Steps to reproduce the behavior:
1. ADD SKETCH (reveals the XYZ planes)
2. Move the pointer to a plane (any of them), and as the pointer rolls over a plane's edge, that plane's edges light up indicating it's ready to receive a click.3. 
3. Any attempt to click misses that the planes disappear.  I usually use a tablet, but with the frustration of tiny hand shakes I cannot successfully select a plane, I changed to a mouse, and still the same.  The planes just drop out of site.
4. No new sketch. 

**Expected behavior**
A clear and concise description of what you expected to happen.
I would expect that the entire face of the planes would be a target for selection to add a sketch.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - OS:  macOS 12.3.1
 - Blender Version: 3.1.2
 - Addon Version: 0.20.0

gitblair, 2022-04-26:
While it is possible to choose a plane, it takes many attempts to get there. 
The hovering that highlights the plane is not consistent. It feels very similar to attempting to choose anything other than the corner blocks. Also similar to the issue where the lines are too thin to hover over and select.

Mac OS 12.2.1
Apple M1
Blender 3.0.0
Addon  0.20.0

hlorus, 2022-04-26:
This is a duplicate, see #29. Closing

closed 2022-04-26 by hlorus