hlorus/CAD_Sketcher #40: [REQUEST] Selection modes

hlorus, 2022-04-26:
**Motivation / Existing Problem**
Native selection tools offer different modes, add the same to the addon's select tool.

**Description**
Add selection modes to allow adding/substracting/toggling selection. Idealy follow blender's default keymap, see: https://docs.blender.org/manual/en/latest/interface/selecting.html#selecting-modes

amrsoll, 2022-08-09:
I guess we should shoot for this selection menu, like when using the Blender selection tool?
[image]

It is not shown when selecting the Solvespace selection tool (yet)
[image]

hlorus, 2022-08-09:
> I guess we should shoot for this selection menu, like when using the Blender selection tool?
> [image]
> 
> It is not shown when selecting the Solvespace selection tool (yet)
> [image]
> 

Exactly! Although i'm not sure if the last two selection modes are really that useful...

hlorus, 2022-10-13:
This is now supported, see:
https://github.com/hlorus/CAD_Sketcher/commit/8b2a71dd6917addfaa891e4c80c43c22054a688d

What's remaining are the shortcut bindings. The problem here is that shift is currently used to select through gizmos.

hlorus, 2022-10-20:
The ctrl and shift modifiers are now also there, constraint selectability was moved to ctrl+shift.