hlorus/CAD_Sketcher #42: [REQUEST] Make Vertical/Horizontal edge constraints harder to delete by mistake

Hyperlooper, 2022-04-27:
First off, I can't tell you how excited I am for this tool. It's something I have wanted for a long long time in Blender! Great work!!

Issue:
Single-clicking to delete the vertical or horizontal edge constraint is too prone to accidental deletions.

I would recommend sticking with Blender UX convention here, and allowing users to select the constraint tickmark and use the standard 'x' keyboard shortcut to spawn the delete modal. If in the future there will be editable params on the vertical and horizontal edge constraints then the delete option could be presented there as well, using the same UX pattern you use for the distance constraint.

hlorus, 2022-04-27:
Constraints are currently displayed with the gizmo api. Having a selection state on gizmos is probably not possible.

The straight forward solution would probably be to always show the gizmo menu, even if there is no property to edit. On the other hand some people might find this annoying, maybe a user preference has to be evaluated.

ZanQdo, 2022-05-03:
IMO this goes for every constraint. Deleting constraints is quite destructive and currently it is too easy to do by mistake.

commit da43bd11 (2022-05-04): Gizmos: Always show constraint menu instead of directly calling delete operator

Before it was too easy to accidentially delete a constraint.
Closes #42
closed 2022-05-04 by hlorus