hlorus/CAD_Sketcher #46: Constraint shortcuts

hlorus, 2022-04-27:


hlorus, 2022-04-28:
@TheMakerTales Would be great to define a list of shortcuts here. Note that these should not colide with the [existing shortcuts](https://hlorus.github.io/CAD_Sketcher/tools/#workspacetools).

TheMakerTales, 2022-04-28:
I think an interesting direction to take the shortcuts might be using taking inspiration from [FreeCAD sketching shortcuts](https://www.scan2cad.com/blog/cad/freecad-keyboard-shortcuts/#:~:text=Shift%20%2B%20F1-,Sketcher%20keyboard%20shortcuts,that%20you%20can%20use%20when%20working%20in%20the%20Sketcher%20workbench.,-Carbon%20Copy)

Here are all the FreeCAD sketches sortcuts 

<html>
<body>

Carbon Copy | C then C (don’t press at the same time)
-- | --
Close Shape | Ctrl + Shift + S
Constrain circle or arc | Shift + R
Copy | Ctrl + C
Connect edges | Ctrl + Shift + K
Constrain Angle | A
Constrain coincidently | C
Constrain Distance | Shift + D
Constrain Horizontal Distance | Shift H
Constrain Vertical Distance | Shift + V
Constrain Equal | E
Constrain Horizontally | H
Constrain Internal alignment | Ctrl + A
Constrain Parallel | Shift + P
Constrain perpendicular | N
Constrain onto object | Shift + O
Constrain Symmetrically | S
Constrain tangent | T
Constrain vertically | v
Create fillet | F
Create Line | L
Create rectangle | R
Extend Edge | T then E
Extend geometry | X
Move | Ctrl + M
Hide/Show internal geometry | Ctrl + Shift + E
Select conflicting constraints | Ctrl + Shift + E
Select constraints | Ctrl + Shift + C
Select elements with constraints | Ctrl + Shift + E
Select the horizontal axis | Ctrl + Shift + H
Select Origin | Ctrl + Shift + O
Select redundant constrains | Ctrl + Shift + R
Select the vertical axis | Ctrl + Shift + V
Toggle construction geometry | C then M (don’t press at the same time)
Trim edge | T then  R (don’t press at the same time)

</body>
</html>

hlorus, 2022-04-29:
Alright so there are a few conflicts with existing shortcuts, to avoid them and to try to be consistent i would suggest to access all constraints through one modifir key e.g. shift. That would result in:

Shift + V vertical
Shift + H horizontal
Shift + C coincident
Shift + P parallel
Shift + N perpendicular
And so on...

TheMakerTales, 2022-04-29:
I think that would be a great idea to go forward.

commit 8696dcdc (2022-05-01): Add constraint shortcuts

Closes #46
closed 2022-05-01 by hlorus