hlorus/CAD_Sketcher #47: Constraints display toggle

hlorus, 2022-04-27:


hlorus, 2022-05-09:
@TheMakerTales not quite sure if this intends to add a global " show constraints" toggle or a per constraint toggle or both...?

TheMakerTales, 2022-05-09:
@hlorus I think that both would be great at minimum individually.
and a little bit of feature creep here but I also have a feeling that a way of hiding all "lines" and all "construction lines" with a toggle would be interesting. 

I say this because while making this there was a lot of manual hiding of construction lines and lines to be sure I was clicking the constraining to the right thing. 

[image]

exegetor, 2022-06-10:
Blender does have a "hide gizmo" button at top of 3dview.  shortcut is ctrl-`
Maybe we just need to put that in the documentation, since it isn't going to be obvious to a new user.

TheMakerTales, 2022-06-10:
@exegetor I have a feeling in the next 2 weeks this might be sorted out tho as it has been commissioned out to a dev I work with quite a bit.

commit 9272629c (2022-08-09): Ui: Display constraints for relevant entities only

Closes #47
closed 2022-08-09 by hlorus