hlorus/CAD_Sketcher #48: Constraints selectability

hlorus, 2022-04-27:


exegetor, 2022-05-20:
can you elaborate on this?   what it would look like and what effects would be possible?

hlorus, 2022-05-20:
@TheMakerTales might know better but as far as i understand it the task is ment to solve the problem of gizmos that obscure entities and therefore disallow entity selection.

IIRC @TheMakerTales proposed to have a toggle which disables gizmos selectability but still renders them.

I'm not sure if this is really needed when we have the constraints visibility toggle as described in #47.

TheMakerTales, 2022-06-20:
@exegetor what hlorus says is correct. It's a way of still seeing the gizmos but making them not selectable as they can sometimes get in the way of more complicated sketches.

commit 6a1a9aff (2022-06-26): Constraints selectability

Allow to select through constraints by holding LEFT_SHIFT

Closes #48
closed 2022-06-26 by hlorus