hlorus/CAD_Sketcher #51: Sketch bevel tool

hlorus, 2022-04-27:


Intuos5, 2022-05-03:
In CAD software, I use the chamfer tool with a distance set to 0 on both ends to let two lines meet at their endpoint. Perhaps such a thing could be part of the bevel tool (considering the bevel tool can do both chamfers and fillets)?

TheMakerTales, 2022-06-29:
I'm been thinking about what this tool could look like and I thought I would just put down a few things in here for food for thought. 

Personally, I really like the Bevel / Fillet tool of Plasticity:
[image]

I was thinking that this is how things could look like once made but the one limiting factor if we hide the constraints is what happens when users want to me "uneven" bevels and chamfers. Should that be part of the tool or a completely different issue as most use cases are looking for even bevels and chamfers...

Personally, I think exposing the constraints driving the chamfer and bevel to be a good thing as this way there is the flexibility to edit and hide them later if you don't want them cluttering up your view.
[image]

[image]

Intuos5, 2022-06-29:
I would prefer to have a clean view of the chamfer/ bevel during its creation, but would like to have the flexibility to edit the constraints afterwards. Ideally, it has a flexible input, so just like Blender's bevel, where you are not forced to type in a radius beforehand. 
For chamfers, distance on both ends are nice, as are angle + distance. I think both methods should then create different constraints. The tool could be modal, so that you can tweak either end through a hotkey press/ modifier hold.
Just some initial thoughts on the matter.

Intuos5, 2022-08-10:
The 0.25 bevel tool looks good! I would like to assign a radius constraint on the fly though. So after picking the two lines to bevel,  I would want to type in a radius, hit enter or click to confirm. Since the value is typed-in, it should also create a radius constraint with the same value. Hope this will alsi be considered 😉

hlorus, 2022-08-10:
Entering a value is already supported, a constraint isn't auto created yet tho. There are some tools that do add a constraint when an exact value is typed in so it's definitely possible.

hlorus, 2022-08-30:
Closed by 7a2a98adbd3f57dcf47c0326151d18fe53d92d76

closed 2022-08-30 by hlorus