hlorus/CAD_Sketcher #58: Entity conversion resolution

hlorus, 2022-04-27:


SupinePandora43, 2022-05-02:
What about `multires` support too?

Cheaterman, 2022-05-08:
Hello! Sorry if this should be obvious, but I don't really understand what this feature is about ; could it be possible to add some additional information, given this is tagged as "good first issue"? I'm thinking "current behavior", "expected behavior", and some pointers as to where in the codebase the change should be made?

It would be awesome if such extra information could be added to other issues with the "good first issue" tag too, please :-) thanks in advance!

SupinePandora43, 2022-05-08:
> Hello! Sorry if this should be obvious, but I don't really understand what this feature is about ; could it be possible to add some additional information, given this is tagged as "good first [...]

There's no way to specify how many polygons sketch-to-mesh result should have (i think)

hlorus, 2022-05-09:
> Hello! Sorry if this should be obvious, but I don't really understand what this feature is about ; could it be possible to add some additional information, given this is tagged as "good first [...]

Absolutely agree with that, thanks for pointing out! :)

hlorus, 2022-06-22:
IMO it's better to have this solved by a generic system rather than hardcoding properties that the user might want to change.

There is already an attempt to solve this, see #200.

hlorus, 2022-08-30:
Closed by fd3b9e3cc30dbbac5a27f3ed840ff1487b40621c

closed 2022-08-30 by hlorus