hlorus/CAD_Sketcher #60: Add radial mode to diameter constraint

hlorus, 2022-04-27:


TheMakerTales, 2022-05-01:
https://github.com/hlorus/CAD_Sketcher/issues/89#issue-1222134997

exegetor, 2022-05-11:
The context menu (shown below) is a good place to select from [diameter, radius, linear distance radius] as well as [inside display, outside) display].
While we're at it, let's restrict the arrow so that it must touch part of the arc being referenced.  (see contrived example below for an "illegal" dimension arrow)

[image]

If there's no rush, I'll give this a shot.

hlorus, 2022-05-11:
> If there's no rush, I'll give this a shot.

Sure thing, if you have questions you can reach out here or on the discord's contribute channel.

exegetor, 2022-05-12:
There are two related issues here.  The first is radius mode vs diameter mode.  This may be an oversimplification, but I think an arc will **always** be dimensioned by radius.  If we check (in _create_shape) `isinstance(constr.entity1, class_defines.SlvsArc)`  we can go ahead and use the radius mode without any user input.  If it's a circle, we would use diameter mode.  I have this much working fine, but it doesn't address FR89.

For that, we need display_mode, which should offer three options: Inside (current style), Outside (Feat. Req. 89, Alt. Sol'n. #2), and Distance-style dimensioning (proposed as FR89, Alt. Sol'n #1.  (Does this have a name?).

Since the helper lines are calculated in _create_shape(), we need to get draw_style over from the context menu (like with Distance alignment setting) to the gizmo.   I'm still struggling with the basics of python and how these Objects communicate state to each other.  If you have time to drop me a hint, maybe point out a similar mechanism already in this codebase and explain the pattern?  

I'm having trouble using discord, but I'll try to get that fixed soon.

hlorus, 2022-05-12:
I would generally advise to try to keep changes small and concentrated rather than trying to implement everything at once. 

Regarding deciding which mode to use: That seems alright, that's also something that could easily be added later if needed. IMO we could even always use the radius mode for now as the diameter mode currently clutters the center.

I would suggest as a first step to add a draw_radius (same as the draw_offset) to the angle constraint. Then use this to extend the constraint line outwards, away from the center. With this you will also have to make sure that this property gets updated correctly when dragging the constraint around. Not sure if this is a bit too much to start with but i think it's an important priority. Anyway i'm here to assist.

Regarding the communication of the draw_style and similar properties: Currently they are stored on the actual constraints e.g. in class_defines.py in the SlvsAngle class. Such properties can be displayed with the class's draw_props method. Inside the gizmo you can access the properties with the constr object (e.g. constr.draw_style).

hlorus, 2022-06-05:
Closing as duplicate of #155

closed 2022-06-05 by hlorus