hlorus/CAD_Sketcher #66: [BUG] NameError: name 'sys' is not defined

callimero, 2022-04-28:
**Description**

Install, then activate. Have seen the same error on 2.93 and 3.0 which I tried also.

Modules Installed (CAD_Sketcher-main) from 'C:\\Users\\Carsten Wartmann\\Downloads\\CAD_Sketcher-main.zip' into 'C:\\Users\\Carsten Wartmann\\AppData\\Roaming\\Blender Foundation\\Blender\\3.1\\scripts\\addons'
Traceback (most recent call last):
  File "C:\Users\Carsten Wartmann\Apps\blender-3.1.2-windows-x64\3.1\scripts\modules\addon_utils.py", line 387, in enable
    mod.register()
  File "C:\Users\Carsten Wartmann\AppData\Roaming\Blender Foundation\Blender\3.1\scripts\addons\CAD_Sketcher-main\__init__.py", line 107, in register
    ensure_addon_presets()
  File "C:\Users\Carsten Wartmann\AppData\Roaming\Blender Foundation\Blender\3.1\scripts\addons\CAD_Sketcher-main\__init__.py", line 97, in ensure_addon_presets
    if sys.version >= (3, 8):
NameError: name 'sys' is not defined

Really Blender?

EDIT: Second activation worked?!
Installed Sover via PIP.
But no sign in Toolbar or N-Panel (sorry old fart user).
Restarted Blender, now it is there.
HOWEVER: If I use my 3D-Printing template with NO objects in the scene, Sketcher is not showing up until restart of Blender. 

**To Reproduce**
Steps to reproduce the behavior:
1. sorry
2. see above

**Expected behavior**

To have it working. Yesterday it worked somehow despite some errors.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - OS:  Win 10
 - Blender Version: 3.1.2, 2.93 
 - Addon Version: 0.20.0

TheMakerTales, 2022-04-28:
Would it be possible for you to send over a blender save in the "3D-Printing template" template?

TheMakerTales, 2022-04-28:
Ok sorted out... It's not to do with the addon it's to do with the "bl_app_templates_user" preferences being used. As the template has a preferences file in there no matter how you setup blender when you go into that template it will over write your new preferences i.e having CAD Sketcker enabled.

If you have no idea what I'm talking about take a look at this video I made about app_templates [App template video](https://youtu.be/8fUqBnNk8KI)

Ultimately all you have to do is follow my video and make some new user preferences for your 3D printing template .

closed 2022-04-28 by TheMakerTales