hlorus/CAD_Sketcher #75: Prompt user to restart blender after installing solver via pip

SupinePandora43, 2022-04-29:


TheMakerTales, 2022-04-29:
I'm not too sure about this one. As it's not always needed. It's very dependent on the user's pip setup and os. Currently, there is also a built-in prompt at the bottom "warring" message letting users know: "Package should be available but cannot be found, check console for detailed info. Try restarting blender, otherwise get in contact."

amrsoll, 2022-08-09:
I think it is a generaly good idea that - if we do already show that warning message - to have a <kbd>Restart Blender</kbd> button.

hlorus, 2024-07-20:
Thanks for the report. With blender 4.20 CAD Sketcher is installed as an extension which brings support for pip dependencies. Blender will install the correct solver module when the extension is installed.

closed 2024-07-20 by hlorus