hlorus/CAD_Sketcher #85: [BUG] Mesh Referencing Workplanes Incorrect Rotation

TheMakerTales, 2022-04-30:
### Description

Workplane rotations not playing nicely with mesh reference workplanes.
[image]

### Addon Version

0.21.0

### Blender Version

3.1.2

### What platform are you running on?

Windows

TheMakerTales, 2022-04-30:
I don't know if this is the same bug some mesh rotation and location is no longer been taken into consideration.
[image]

closed 2022-05-01 by hlorus

hlorus, 2022-05-02:
Fixed by: 8e77fe00807a0bd337d6b53e7f55cc369bfd8710