hlorus/CAD_Sketcher #89: [REQUEST] Change how diamater constriant is displayed

Maykeye, 2022-05-01:
**Motivation / Existing Problem**

Now diameter constraint has text inside the circle:
[image]

Constraint line and text completely cover the center point. I can't do anything with it.
If I try to draw a new line from the center point, I can't, at least every time I try, it prompts to change/delete constraint, even though I use tool to draw 2d line, not selection tool:
[image]

**Description what should happen**
I should be able to choose center point without pixel hunting for it. Right now we can move distance constraint around. We can only rotate diameter constraint around the center, but it rotates constraint line without moving the text.

**Alternative Solutions**
1) I find it's easier to not use diameter constraint at all and instead create vertical/horizontal line and add distance constraint=radius of circle
[image]

2) I can try to start line from another point and land near center, hoping that the center will be selected, but most of the times I create new point instead, but it wouldn't work if I want to draw a line between two circles with constrained diameter

**Mockup**
* Add ability to move text out of the circle, so center point can be selected, like this
[image]

* If user clicks near center point, blender should prioritize selecting center point, not constraint.

TheMakerTales, 2022-05-01:
Hi Maykeye,
Thanks for the detailed request. I'm happy to say that something like this is already planned: [https://github.com/hlorus/CAD_Sketcher/issues/60](https://github.com/hlorus/CAD_Sketcher/issues/60)

closed 2022-05-01 by TheMakerTales

deathbypain, 2022-05-10:
To add on to this, if two circles that share a center point both have diameter constraints, the text for the constraint display overlaps and is illegible.

[image]