hlorus/CAD_Sketcher #93: Keeping Material & Shading properties after Sketch edits

TheMakerTales, 2022-05-01:


hlorus, 2022-05-01:
I've made a change to not always create a new mesh data block, this should help for properties that are stored on the mesh. However it's another story for data that is stored on mesh elements as they get overwritten. This is somewhat related to #6 as existing data would then only be updated...

TheMakerTales, 2022-05-01:
shading might not be supported but wow, materials sorted plus that also added support for linked duplicates

TheMakerTales, 2022-05-04:
Shall this issue now be closed and a feature request for smooth shading support?

hlorus, 2022-05-04:
> Shall this issue now be closed and a feature request for smooth shading support?

Yeah i'd say that makes sense.

TheMakerTales, 2022-05-04:
Smooth Shading Support moved to feature requests https://github.com/hlorus/CAD_Sketcher/discussions/104

closed 2022-05-04 by TheMakerTales