hlorus/CAD_Sketcher #103: Incorrect highlighting of constraints [BUG]

ZanQdo, 2022-05-03:
### Description

Mouse over highlighting seems to have problems in my system

Operating system: Windows-10-10.0.19044-SP0 64 Bits
Graphics card: NVIDIA GeForce GTX 1080 with Max-Q Design/PCIe/SSE2 NVIDIA Corporation 4.5.0 NVIDIA 512.59

Here is a video demonstration of the problem where highlighting is wrong (notice the position of the mouse cursor and which constraint gets highlighted)

https://user-images.githubusercontent.com/999476/166588333-9455a2fc-8932-428c-ae09-21a6854b89e2.mp4

### Addon Version

0.22.2

### Blender Version

3.3.0

### What platform are you running on?

Windows

hlorus, 2022-05-04:
Thanks for reporting, could you share the blend file?

ZanQdo, 2022-05-04:
Attached test file to the original post

hlorus, 2022-05-10:
Alright, bisecting led to https://developer.blender.org/rB9dadd8bdb101fd5c06cd6ca7aa708a6ce6f13d47 as the first bad commit. Going through the tasks it looks like this has been reported as a high prio bug, see:
https://developer.blender.org/T97771

Let's hope this gets fixed :)

ZanQdo, 2022-05-10:
Ah, no wonder, it seemed to have appeared in the latest versions indeed. That Campbell! :)

hlorus, 2022-05-11:
This should be resolved as the causing commits have been reverted.

closed 2022-05-11 by hlorus