hlorus/CAD_Sketcher #110: [BUG]Missing Platforms

denverh2000, 2022-05-07:
### Contact Details

denverh@comcast.net

### Description

Besides Windows, Mac, and Linux systems, there are also the BSD systems that can use Blender.  Additional platforms need to be included in get_wheel() in preferences.py.  

### Addon Version

0.22.1

### Blender Version

3.0.0

### What platform are you running on?

Linux

hlorus, 2022-05-09:
Hey, thanks for the hint. The platforms are limited by the availability of the [solver module](https://pypi.org/project/py-slvs/#files). It doesn't seem like there are additional .wheel files for such platforms, i'm not sure if the module is actually supported there.

hlorus, 2022-05-10:
Anyway the get_wheel function is only used to detect the solver module and set it as default if its bundled with the addon which isn't currently used. Installing a .whl file is possible on all platforms.  Closing..

closed 2022-05-10 by hlorus