hlorus/CAD_Sketcher #120: base the amount of dimension-line overshoot on the arrow size

exegetor, 2022-05-09:
An overshoot is calculted for the extender lines for dimensions.  The code is in gizmos.py,
and around line 280 and 370 we find:
        overshoot = math.copysign(0.04, offset)
and
        overshoot = math.copysign(0.04, radius)

I found this overshoot value way our of proportion on a small part (<10mm).  I've played around at different scales and found that

        overshoot = math.copysign(GIZMO_ARROW_SCALE * 0.02, offset)
and
        overshoot = math.copysign(GIZMO_ARROW_SCALE * 0.02, radius)

provides a pleasing appearance.

Cheers, and great work on this!  I haven't been so excited for an add-on since dynamic paint!

hlorus, 2022-05-09:
Hey thanks for the constructive input! Since you've already figured out how to improve this would you like to submit a Pull Request? Otherwise, if this is too big of a hassle i can of course do a commit for you.

exegetor, 2022-05-09:
well, on further consideration, i haven't really solved anything.  let's junk this patch and discuss how to do it properly.
here's three ideas---
1. Make the offset exactly the size of the arrowhead line.  It looks like the arrowhead dynamically sizes with the 3dview zoom level to give a "good appearance" as you zoom in/out.   Matching that might work.  Sadly, I don't understand the code / API enough yet to get that right :/
2. Add Scaling Factor to Preferences/General (alongside Entity Scale, Icon Scale, and Text Size)
3. Go back to zero offset, and have the lines end at the terminus of the arrowhead.  I saw it in this state in a youtube video and it looked good that way!

hlorus, 2022-05-10:
Get_scale_from_pos is doing the whole magic here, this already gets called for the arrow positions. It should be sufficient to scale the offset with that but maybe there's a bit of reordering needed so that function is only called once per arrow position.

hlorus, 2022-05-11:
Hmm for some reason i cannot apply your patch, git yells at me: error: "No valid patches in input".

How did you create the diff?

exegetor, 2022-05-11:
Badly!  Backwards, in fact
[gizmos.py.diff.txt](https://github.com/hlorus/CAD_Sketcher/files/8667210/gizmos.py.diff.txt)

commit a6ecd1df (2022-05-11): Gizmos: Scale line overshoot

Inital patch by @exegetor
Closes #120