hlorus/CAD_Sketcher #126: Invert the angle value when inverting an angle constraint [REQUEST]

ZanQdo, 2022-05-12:
**Motivation / Existing Problem**
Let's say we are constraining the external angle to 45 degrees
[image]

When inverting the constraint to the inside. Those 45 degrees are no longer applicable
[image]

**Description**
A clear and concise description of what you want to happen.

**Alternative Solutions**
A clear and concise description of any alternative solutions or features you've considered.

**Mockup**
Ideally add a quick drawing to show how this would look.

exegetor, 2022-05-12:
I might have found the problem.  I think the Invert setting is checked too early and a new angle (135) is sent to the solver, which needs to find new geometry for the "new angle".   I would leave the solver data alone and have the Gizmo check the Invert setting and draw helper lines accordingly.  If that sound like a good plan, I can get that done tomorrow night.

hlorus, 2022-05-12:
> I might have found the problem.  I think the Invert setting is checked too early and a new angle (135) is sent to the solver, which needs to find new geometry for the "new angle".   I would leave [...]

Not exactly sure what the right approach would be here, just keep in mind that the solver method "addAngle" takes an invert option as input, therefore i guess it makes sense to pass the constraints invert setting to the solver.

It might make sense to modify the constraint value from the invert setting's update callback.

exegetor, 2022-06-08:
I've been having a bit of "technical difficulties"... using Visual Studio Code, if I add anything to update_system_cb(), it  triggers
some kind of error when it hits the callback function... the "def" line.  Breakpoints stop working also.  Possibly somthing wrong in the IDE?

```
[class_defines.py, from line 1890]
def update_system_cb(self, context):
    if hasattr(self, "label"):
        if self.label == "Angle":
            self.value = math.pi - self.value
    sketch = context.scene.sketcher.active_sketch
    solve_system(context, sketch=sketch) 

```
The error message is useless, and occurs numerous time.  No traceback, just a reference to the function definition line:
```
Traceback (most recent call last):
File "C:\Users\shann\AppData\Roaming\Blender Foundation\Blender\3.1\scripts\addons\CAD_Sketcher-main\class_defines.py", line 1890, in update_system_cb
```

Is there something wrong with the python syntax?  (of course, the two if statements can go on one line, and who knows if the logic's good, but damn, I can't even debug it!

exegetor, 2022-06-12:
I put up a PR before noticing the different behavior between the pop-up that appears when you initially set the constraint, vs the context menu Invert setting for an Angle already in place.  The 2nd works great, but the 1st  (seemingly) doesn't apply valve = pi - value, so the geometry jumps accordingly.  I did not forsee that!

commit 9e4eaf32 (2022-06-24): Merge branch 'exegetor-angle_inv'
by @exegetor
closes #178, closes #126
closed 2022-06-24 by hlorus