hlorus/CAD_Sketcher #128: [REQUEST] add option to draw dimensions inside or outside the ledger lines

exegetor, 2022-05-13:
All three styles of explicit dimensions (distance, angle, diameter) have traditionally been drawn in Inside and Outside forms.  Implementing Outside draw_mode for Diameters immediately solves #89 (diameter dimension interferes with access center point of circle).  

This work is along the lines I've been looking into, for example #126  (the Angle Inversion) and #60 (radius mode for arc/circles).

[image]

exegetor, 2022-05-16:
I have Diameter finished (for some values of finished)
The same method (different math tho) should get similar behavior from the Distance and Angle constraints

https://user-images.githubusercontent.com/10360161/168536654-db85506d-a476-4f01-96e4-28dfbaf3a326.mp4

hlorus, 2022-05-16:
IMO it would make sense to submit a patch for this alone as it's probably the most asked for bugfix/feature request.

Btw the dimension line could probably be removed from the selection area as it always occludes the centerpoint of arcs/circles.

exegetor, 2022-05-16:
I created a PR that represents the state of things in the video I posted.  I'm not quite sure what your suggestion ought to look like, and I don't want to trip over my feet, so I'm going to leave that issue for later.  I would hope we could handle the occlusion of the centerpoint by prioritizing selecting entities over constraints, but you'd know best.

closed 2022-07-23 by exegetor