hlorus/CAD_Sketcher #131: Align View with sketch [REQUEST]

mikemac30, 2022-05-16:
**Motivation / Existing Problem**
All cad programs I've used in the past automatically align the viewport camera to the current sketch. Usually you can still move the view around but by default they typically aligns the view to be normal to the sketching plane. This makes editing the sketch easier because the sketch isn't warped by perspective.

**Description**
After clicking "Add Sketch", or the edit icon "Set Active Sketch", the blender viewport should align itself to be normal to the sketching plane.

hlorus, 2022-06-04:
Id suggest to have a focus-sketch operator and a preference to auto execute that on sketch enter.

hlorus, 2023-05-12:
This is already implemented but hidden under an experimental setting. To activate it set the "Show Debug Settings" in the addon preferences, then activate the "Align View" setting.

There are still some visual glitches during or after the transition, at least on my setup. The code can be found here:
https://github.com/hlorus/CAD_Sketcher/blob/main/operators/utilities.py#L85

It would probably also make sense to put the functionality into it's own operator.

kaiwas, 2024-10-13:
I'm looking forward to the release of SketcherCAD. With this tool, Blender's capabilities will greatly increase.

Auto-alignment is a very useful thing, but I hope that there will always be an option to turn it off.

Off-topic question - maybe you know how to turn off auto-alignment in FreeCAD? (I think you are involved in the development)
It's just awful that there is no way to turn this option off.

hlorus, 2024-10-13:
> @hlorus  Off-topic question - maybe you know how to turn off auto-alignment in FreeCAD? (I think you are involved in the development)
> It's just awful that there is no way to turn this option off.

No clue if it's possible to turn that off. I'm involved with the development of the measure facility only so i cannot really help with that, sorry.

kaiwas, 2024-10-13:
> i cannot really help with that, sorry.

The measure facility out great!
As for the question - nothing to worry about. No need to apologize ;).
Thanks for the answer! I wish you success!

commit 9c45b6f2 (2024-08-20): Ops: Add Align View operator

This adds an operator to align the view to a given sketch or back to the default view orientation. It's triggered by:
- Shortcut "V"
- Activating a sketch
- Leaving a sketch

Closes #131