hlorus/CAD_Sketcher #143: [REQUEST] Entity scale for dimension lines in the addon's preferences

Intuos5, 2022-05-27:
**Motivation / Existing Problem**
On my 27inch 4k monitor, I find the dimension lines a bit too thin, especially when they overlap with the grid (overlay).

**Description**
If there was an dimension line entity scale, this would already improve.

**Alternative Solutions**
I have changed the default colour to cyan, this helps improve the situation a little. 
[image]

hlorus, 2022-06-14:
There's a (line width) property in the gizmo api which doesn't seem to work unfortunately. Also as it would be great to have the actual dimension line thicker than the helplines, a different solution is needed.
An approach could be to draw rectangles instead of lines.

exegetor, 2022-06-19:
The gizmo api didn't work but the gpu api still works.  I figure this could be enhanced so the leader lines > witness lines... but that would involved a lot.  I thought I'd just show this and see how you feel about proceeding.
   
https://user-images.githubusercontent.com/10360161/174460785-4036e4ad-ac5d-428e-ba05-0e937013654e.mp4

Intuos5, 2022-06-19:
Do you guys report any of the APIs that are not working correctly as bugs? In which case, it may be good to post the bug report link so people know why it is held up (and can see when they can work on it again). 

Looks like what I want @exegetor 👍

hlorus, 2022-06-19:
I'm afraid the gpu linewidth isn't supported on macs... Not sure if the property in the gizmo api works on another level such that it would be supported on all platforms.
I haven't yet checked if it's supported in blender's issue tracker.

Afisa46, 2026-10-02:
Hi @hlorus. I see this issue is assigned to DariusJDebugs, so I do not want to duplicate their work. Is the assignment still active, or would you consider another contributor if the current assignee is no longer working on it?

If it becomes available, could you confirm the current Small-tier bounty amount, whether PayPal payment is supported, and whether an AI-assisted implementation/review/testing workflow is eligible with clear disclosure? Which Blender versions and platforms would need to pass validation?

This is an availability and eligibility inquiry only. I have not claimed the bounty or started implementation, and I am not committing to a delivery date. Thank you.

agustinbracamonte, 2026-10-05:
Hi @hlorus @DariusJDebugs — checking whether this Small bounty is still actively being worked. If the assignment can be released, I’d like to take it on and deliver a PR. Happy to follow the bounty board payment rules (PayPal/Wise). Thanks.

hlorus, 2026-10-06:
It's free now, the previous assignee has been inactive.

@agustinbracamonte you asked to take it on, so it's yours if you still want it: say the word and I'll assign you. @Afisa46 thanks for checking first.

`Bounty - Small` is $100, paid through our Open Collective; PayPal is fine. AI assistance is fine with disclosure. CI covers Blender 5.0 and latest on Linux; we ship for Windows, macOS (Intel and ARM) and Linux.

One pointer: the old blocker here was `gpu.state.line_width_set` not working on macOS. The codebase now uses the `POLYLINE_*` shaders with a `lineWidth` uniform (see `gizmos/workplane.py`), but the dimension gizmos still build a `"LINES"` shape (`gizmos/distance.py`), so that is where the work is.

agustinbracamonte, 2026-10-06:
Thanks @hlorus — yes, please assign me. I'll switch the dimension gizmos in `gizmos/distance.py` to the `POLYLINE_*` shaders with a `lineWidth` uniform (following `gizmos/workplane.py`) and add a scale preference. I'll disclose AI assistance in the PR. PayPal via Open Collective works for me.

agustinbracamonte, 2026-10-06:
Thanks a lot for assigning it to me, @hlorus! I'm on it and will open the PR soon.

closed 2026-10-06 by hlorus