hlorus/CAD_Sketcher #144: [REQUEST] Option to delete a sketch by right clicking on the Sketch name in the N-panel

Intuos5, 2022-05-27:
**Motivation / Existing Problem**
Currently, there seems to be no way to delete a sketch if it is not converted to a Mesh or Bezier Curve. 

**Description**
To solve this problem, I propose to add a right click option in the N-panel to remove a sketch. 

**Alternative Solutions**
Alternatively, hitting the delete button in the Sketches list should also delete a sketch.

hlorus, 2022-05-28:
You should be able to delete any active sketch, there's a button in th sidebar,right? It would be great though if a sketch could be deleted if it's not active but i'm afraid it's not possible to use right click on a sketch in the ui list.

Intuos5, 2022-05-28:
Could there also be a button to delete any non-active sketches? So select a sketch and hit the button or use a hotkey?

hlorus, 2022-05-31:
Not sure about using a hotkey but maybe just show a X for the active list element.

Intuos5, 2022-05-31:
Just FYI I tend to hotkey everything if I can, an X is fine by me.

commit 555ba240 (2022-06-04): Ui: Show delete operator in sketch ui-list for active element

Closes #144
commit edcff581 (2022-06-27): Ui: Show delete operator in sketch ui-list for active element

Closes #144
closed 2022-06-04 by hlorus