hlorus/CAD_Sketcher #146: [REQUEST] Flip dimension text when upside down instead of aligning the text to the screen

Intuos5, 2022-05-27:
**Motivation / Existing Problem**
Currently, the dimension text rotates aligned to the screen. This causes the text to overlap with the dimension lines, which makes it harder to read. 

**Description**
Instead, the text should flip up-side down when the view rotates in a direction that the text is close to an angle of 180 degrees relative to the regular reading angle. 

**Mockup**
Uploading Rotate dimension text.mp4…

This mockup also applies to #145

hlorus, 2022-05-28:
I'm not exactly sure about this, i'd say visual clutter is still possible even if it's aligned with the dimension line as it can overlap on the sides. I think there will be less clutter when it will be able to drag the constraint text away from the constraint which @exegetor seems to be working on, see: https://discord.com/channels/882152824249667594/967316874121588777/976216964043272242

Intuos5, 2022-05-28:
In that case, you should try using Fusion for a few minutes. The difference aligned text makes to reduce visual clutter is bigger than you may expect ;)
By itself it is indeed not sufficient, but together with dragging dimensions out, I'd say it is.

hlorus, 2022-05-31:
> In that case, you should try using Fusion for a few minutes. The difference aligned text makes to reduce visual clutter is bigger than you may expect ;)
> By itself it is indeed not sufficient, but together with dragging dimensions out, I'd say it is.

I've tried and also tried inventor(where it's view aligned) and didn't really have problems with either one. 
I think there is a tradeoff between clutter and readability, having it view aligned is nice for reading the text but might overlap more.

Another pro is that it's just more performant to draw text in screen space.