hlorus/CAD_Sketcher #153: [REQUEST]Pie Menu

Myrkskog, 2022-05-28:
Extensive Pie Menu that can replace completely the use of T-Panel and N-Panel.

Below is a layout with keymaps that have worked well in my testing. Feels speedy and intuitive.

[image]

Above was created using Pie Menu Editor. Experiencing trouble obtaining the behavior expected for activate sketch / leave sketch portion of pie menu. This particular portion of the pie should invoke solvespace select and active sketch. Upon "leave sketch" it will revert to select tool outside of solvespace and deactivate sketch. Leave sketch works simply given index is always -1.

Currently, sketches can cycle from 0 onward easily. Renaming the sketch works through the pie menu as well.
Some difficulties exist when selecting convert type.

I can provide what I have for a jumping off point should you be interested to add a pie menu. Or at least to test functionality of this pie menu design.

therealjacobbrauer, 2022-09-01:
@Myrkskog I'm interested in adding a pie menu fulfilling hlorus's 'good first issue' with this. @Myrkskog would you mind providing the code for this? Or where should I look for the code for this in what specific fork?

Myrkskog, 2022-10-03:
@therealjacobbrauer this was made with pie menu editor 1.18.4 by roaoao to test how practical and natural it felt for use with CAD Sketcher. I can provide the .json of which I cannot say any useful code can be derived. If you have the addon then you can test the pie with this file. WIP and have not tested this in a while.
[CAD_SKETCHER_PIE_0.0.6.json.zip](https://github.com/hlorus/CAD_Sketcher/files/9697684/CAD_SKETCHER_PIE_0.0.6.json.zip)

closed 2022-10-03 by Myrkskog

Cheaterman, 2023-02-14:
Thanks for the JSON!

> Experiencing trouble obtaining the behavior expected for activate sketch / leave sketch portion of pie menu. This particular portion of the pie should invoke solvespace select and active sketch. [...]

While it probably will be a lot easier to understand once I "get my head into the game" so to speak, could anyone attempt to clarify what @Myrkskog meant here, just to make sure we're all on the same page? Thanks in advance :-)