hlorus/CAD_Sketcher #155: [REQUEST] Dimension Arcs by Radius; Circles by Diameter

exegetor, 2022-05-30:
It is traditional to use diameter for holes (circles) and radius for fillets and the like (arcs).
For clarity, we can prepend an "R" to radii.

When drawn inside, the circle dimension line should go all the way across, but the radius should not.
Similarly, when drawn outside, the circle should have a two-part gizmo with a second arrow diametrically opposed.

[image]

exegetor, 2022-05-30:
PR #156 is up and ready for code review.   My understanding of a class's properties vs. regular variable is uncertain.  I am playing "monkey see monkey do" and hoping I've gotten it right.
Take a look at the way I set up the 2-part gizmo.  Am I missing a more elegant way?  I needed to use hasattr(coord2), which is also a place where I might screw up from lack of experience.  Seems to work, but feel free to tell me I code like a kindergartener :)

hlorus, 2022-05-31:
Absolutely agree with having a radius mode

exegetor, 2022-06-05:
I've added the context menu control to select radius v. diameter  (default set by arc/circle type)
I'm having trouble making the constraint value depend on radius mode, however.  Attempts yielded nightmare code.
So, as it stands, gizmo.py will draw the gizmos and display the value as required, but it's still only on the "presentation" layer.
It's not that I necessarily disagree with you on this point, but.... it this way so wrong? And if so, perhaps I need help making it work as you describe.

this represents the current state of the PR:

https://user-images.githubusercontent.com/10360161/172031751-74fa16ce-c135-465c-8274-81f37605d409.mp4

closed 2022-06-10 by exegetor