hlorus/CAD_Sketcher #160: Constraint Icons in N-Panel

hlorus, 2022-06-04:
### Discussed in https://github.com/hlorus/CAD_Sketcher/discussions/122

<div type='discussions-op-text'>

<sup>Originally posted by **coleorloff** May 10, 2022</sup>
**Motivation / Existing Problem**
It's a little difficult to "read" sketches without familiarity with the constraint icons. (It's also a Blender UI convention to have icons paired with menu items.)

**Description**
Add constraint icons next to their name in the CAD Sketcher N-panel. T

his would provide a positive feedback loop in terms of visually identifying symbols within the system and "reading" your sketches, both for beginners and more advanced users.

**Mock-Up**
Note: highlighted area shows a selection of constraints paired with their icons. The goal would be to have all constraints in the menu paired with their icon. This is only for illustrative purposes. 

[image]

</div>

hlorus, 2022-06-04:
That would be great, lets try to have this for 1.0

hlorus, 2022-06-14:
A few notes: The icons that are currently drawn are blender geometry icons, same as they are used in the toolbar. Theses types of icons cannot be drawn in the layout.
I suggest to look at the documentation [here](https://hlorus.github.io/CAD_Sketcher/)
There are some people in the discord that are willing to help out with icons, best to ask there for help.

Mortal-Engine, 2022-06-14:
Hi! Whats the icon size that you need? 128x128 seems to be food for KDE desktop icons

Mortal-Engine, 2022-06-15:
[image]

[image]

Would these types of icons work? There are taken directly from FreeCAD and recolored to match the CAD Sketcher theme

amrsoll, 2022-06-26:
I think it would be good to have a variety of colors to better distinguish them, or have them follow the sketcher theme colors (which can be modified by the user)

hlorus, 2022-06-28:
IMO having different colors isn't a good idea as it kind of goes against blender's mentality to avoid a rainbow UI, better just have one accent color and maybe use this to communicate something like it's used for blender's workspacetools (where green is used as "add" and red as "remove" etc.).

I wouldn't recommend bothering to convert the existing icon set, they aren't well made and we don't need both types of icons, better to just support drawing regular interface icons in the viewport.

Cheaterman, 2023-02-14:
I suppose this issue will be best used for tracking code progress on implementing the new layout for now, and another issue could be made for artists to suggest new icons? My thinking goes, once the layout is implemented, changing assets should be trivial?

commit 802ee96e (2023-02-17): Add icons to "Add constraints" panel, refactor icon_manager accordingly

Closes #160
PR #337 (2023-02-17): Add icons to "Add constraints" panel, refactor icon_manager accordingly
Closes #160
closed 2023-02-17 by Cheaterman