hlorus/CAD_Sketcher #161: Circle and curve tangential dimension

hlorus, 2022-06-04:
### Discussed in https://github.com/hlorus/CAD_Sketcher/discussions/112

<div type='discussions-op-text'>

<sup>Originally posted by **hbfortunato** May  7, 2022</sup>
When using any parametric modeling software, normally you can do measurements like this
[image]

The existing problem with the addon is that you can only do dimensioning only through vector points. whereas shown in the image it doesn't refer you to a vector point and always finds the accurate nearest side of the circle to the line.

here are the sample from the addon
[image]

ALTERNATE SOLUTION:
The alternate solution for this is to create a tangent line of a circle so that it can measure from the vertex of the reference tangent line up to the line it will be measured.
[image]

The problem with that is that it is too tedious to work with. Also, I think the main problem in line with this, for now, is the dimensioning, wherein you are always doing dimensioning from point to point,

[image]

That is why when it comes to tangential dimensions from circle/curve to line won't work.

Lastly,

If we can have a solution that we can measure or add a dimension not just only from the vertex/point only. it would be very beneficial for the user.

[image]

I know you are in a very early stage but I hope this will help many in the near future.</div>

hlorus, 2022-06-04:
Let's try get that for 1.0!

TheMakerTales, 2022-06-29:
This issue has been given a $200 bounty!
Find out more about bounties here: [https://www.cadsketcher.com/cad-sketcher-bounty-board](https://www.cadsketcher.com/cad-sketcher-bounty-board)

hlorus, 2022-08-30:
This is now implemented by exegetor: https://github.com/hlorus/CAD_Sketcher/commit/07362c2affc993c99e873137f88827c3ba1cb3d6

closed 2022-08-30 by hlorus