hlorus/CAD_Sketcher #162: [BUG] Construction circles not visible

BrCadfael, 2022-06-04:
### Contact Details

policebox@charter.net

### Description

When I set the Construction check on a circle, it disappeared.  Turning off the Construction check restores the circle.  I verified that the visible check was set the whole time.  My guess is that it doesn't know how to draw a construction circle.

### Addon Version

0.22.6

### Blender Version

3.1.2

### What platform are you running on?

Windows

BrCadfael, 2022-06-04:
Additional information.  The circles do appear if you zoom into them.  The problem appears to be connected to the size of the circle on the screen.  A large construction circle will be visible, a smaller one may not be.  There is a transition point where the circle "materializes" as you zoom in and "dematerializes" when you zoom out.  On my screen that happens when the circle is about a centimeter across.  The issue doesn't appear to affect lines at all.  Construction lines are always visible.

hlorus, 2022-07-03:
Looks like a glitch with the opengl dashed shader. I can reproduce this but it's usually not a problem for the scale i'm working with.

amrsoll, 2022-08-09:
For some context
[image]

[image]

[image]

jvenborg, 2022-08-31:
Yep, that has become an issue for me as well. Solid lines in circles are all fine, but construction lines will become invisible. In my current sketch, the diameters of the circles are 2 and 5 mm, respectively. Version 0.25.0.

Miguel-Rodrigues, 2023-06-04:
Same thing happens here:

[image]

[image]

Windows 10
Blender 3.5.1
CAD Sketcher 0.27.1

closed 2024-08-27 by hlorus