hlorus/CAD_Sketcher #191: [BUG] Vertical/horizontal distance constraint causes jumping

BaoNTGH, 2022-06-16:
### Description

Repro step:
1. Add sketch
2. Add line
3. Select 2 end points of the line
4. Alt-H (horizontal) or Alt-V (vertical)

Expected:
Horizontal or vertical constraint is added with current distance value without moving the points of the line.

Actual:
Different distance value was provided that caused the line jumps
[image]

[image]

### Addon Version

latest

### Blender Version

3.1.2

### What platform are you running on?

Windows

TheMakerTales, 2022-06-18:
Ya I've experienced the same when sketches have zero or lake constraints

amrsoll, 2022-06-29:
It looks like the horizontal/vertical distance constraints use the length of the line as the value for the constraint, instead of using the projection of the line onto the given axis for the measurement.

commit 1de5d23b (2023-03-02): Fix: Vertical/Horizontal constraint causes jumping

Fixes #191
Also solves weird behaiviour in workplanes other than XY
closed 2023-03-02 by hlorus