hlorus/CAD_Sketcher #196: [BUG] On discord server, when I react thumbsup to the post, it doesn't work.

SomeDevWeb, 2022-06-20:
### Contact Details

somedev@gmx.com

### Description

This is just a Discord (more of usability) bug, but nevertheless, it blocks the user access and thought I should notify you.
So when I joined your server, there is a message that "Once you have read... react 👍 to this post to grant membership"
The issue is that at some point I seem to have lost the default (yellow) thumb icon in discord and I just have the whiter one (with code :thumbsup_tone1:). 
I reacted with it and nothing happened, so I thought the code only accepts the default one. 
Therefore, I'm actually stuck cause I can't do anything about it. I tried multiple times to restore the default thumbnails, but nothing worked. But I think the server should accept the other thumbsup icons as well.
BTW, as a thought I saw there are a few other that reacted with the same icon, but I don't know if theirs were accepted or not.

### Addon Version

0.20

### Blender Version

3.2

### What platform are you running on?

Windows

exegetor, 2022-06-21:
It's not asking for you to send an emoji by chat, but to click on the existing 769 reactions to add yourself to that number

SomeDevWeb, 2022-06-21:
It did. 🙂
O man, I must be either not too bright, or that message is completely missguiding saying you need to react to that message.
Anyway thanks for clarification. :)

closed 2022-06-21 by hlorus