hlorus/CAD_Sketcher #213: [BUG] constraints invisible after upgrading to 0.24.0

dallinbeutler, 2022-07-02:
### Description

I was using the previous release. I tried installing the new 0.24.0 release and upon enabling the plugin I get this error:
[image]

If I click the arrow again the plugin loads, but no constraint lines are visible. happens with a new or existing sketch file

### Addon Version

0.24.0

### Blender Version

3.2

### What platform are you running on?

Windows

hlorus, 2022-07-02:
Looks like it is failing when reloading the scripts... Do you get the error also when the addon registers at startup?

amrsoll, 2022-08-09:
Very interesting to see Blender is available on the steam store :monocle_face: :laughing: 

Otherwise, this usually points to circular imports. It seems to be caused by some black (grey?) import magic. 

@hlorus do you remember why it is that these dynamic import statements exist? Could it be we need to reorder the `importlib.reload()` statements ?

https://github.com/hlorus/CAD_Sketcher/blob/690539fb8fef1c087049d46e7bcba3f5a7450535/__init__.py#L14

If I am correct, then **re-installing the plugin should fix the issue**.

I think Blender is trying to live-reload the plugin after the update and there are a few caveats that might not be working properly.

hlorus, 2022-08-09:
Importlib.reload() was used to support the reload_scripts operator which allows to update the addon while blender is running. This has been changed however and is now done by cleaning sys.modules.
If the issue here is provoked by reloading, which seems to be the case, it's likely resolved.