hlorus/CAD_Sketcher #214: Blender crashes to Desktop

hrjrkr, 2022-07-02:
Hi
 After I have build a sketch and try to export it as a pdf, Blender crashes to Desktop. Not sure if it's Blender, CAD Sketcher or me.
Thanks
Joe

hlorus, 2022-07-03:
How would you export to pdf? I assume you're converting a sketch to mesh and then export that somehow, right? Is there an addon involved to do that?

hrjrkr, 2022-07-03:
Hi
 No, I did not convert it to mesh. I just wanted a pdf of the sketch. In my line of work, I draw a lot of sketches. I guess I could just do a screen shot.
Thanks
Joe

hlorus, 2022-07-03:
I don't think the crash is related to the addon then. Also i wouldn't expect that there would anything show up in the pdf as tha addon isn't using a native geometry type.

closed 2022-07-03 by hlorus