hlorus/CAD_Sketcher #216: [BUG]

mac4kent, 2022-07-04:
### Contact Details

Dark, Light bug on Edit - Preferences on Add-on tab 

### Description

[image]

Dark and Light bug not work on Blender 3.2.0 and 3.2.1
[image]

then next click Dark
[image]

### Addon Version

0.24.0

### Blender Version

3.2.0 and 3.2.1

### What platform are you running on?

Mac OSX

hlorus, 2022-07-05:
Did you already install the solver module whwn this error happened?

mac4kent, 2022-07-05:
let me check if your update time/date.

mac4kent, 2022-07-05:
did not resolve. I already click install button yesterday same time 0.24.0 add-on yesterday (July 4, 2022)
[image]

mac4kent, 2022-07-05:
My screenshot record. https://youtu.be/9XgesKLgcks you can see. I am not good english explain text.

hlorus, 2022-07-05:
Could you check the console for any messages? https://hlorus.github.io/CAD_Sketcher/advanced/#console-output

mac4kent, 2022-07-05:
Linux/Mac is not support because of Linux is a Windows. Mac is not Windows.
I can tell C:/ is not support on Mac only.
Mac is not support (CTRL+ALT+T) never support.
AppData not there.

You give me link there many is not match on Mac. They said (Linux/Mac) is not support. 
Must separate (Windows / Linux / Mac) because Mac is very special direct not Linux.

[image]

[image]

[image]

mac4kent, 2022-07-21:
I am wait for you answer

hlorus, 2022-07-22:
Could you try with the latest version from github?

mac4kent, 2022-07-22:
I still can't read black line plane floor after click.
https://youtu.be/zCarMBGwUwE

amrsoll, 2022-08-09:
Hey @mac4kent

On your last video, everything looked mostly normal. ( The "solver Module" did not install correctly though)

We know that Mac an Linux are very different platforms, but the program is mostly OK for both platforms. Are you able to start your Blender program in a terminal? 

> The `AppData` example is just an example, it could look different for you

When you select the workplane, they disappear. This is normal. You can now start making lines, circles etc... Your interface is original, and I cannot see the buttons, but here are the shortcuts:

> https://hlorus.github.io/CAD_Sketcher/tools/

|Key|Modifier|Action|
|:---:|---|---|
|ESC|-   |Activate Tool: Select|
|P|-   |Invoke Tool: Add Point 2D|
|L|-   |Invoke Tool: Add Line 2D|
|C|-   |Invoke Tool: Add Circle|
|A|-   |Invoke Tool: Add Arc|
|R|-   |Invoke Tool: Add Rectangle|
|S|-   |Invoke Tool: Add Sketch|
|Y|-   |Invoke Tool: Trim|

**Dimensional Constraints:**

|Key|Modifier|Action|
|---|---|---|
|D|Alt   |Distance|
|V|Alt   |Vertical Distance|
|H|Alt   |Horizontal Distance|
|A|Alt   |Angle|
|O|Alt   |Diameter|
|R|Alt   |Radius|

**Geometric Constraints:**

|Key|Modifier|Action|
|---|---|---|
|C|Shift   |Coincident|
|V|Shift   |Vertical|
|H|Shift   |Horizontal|
|E|Shift   |Equal|
|P|Shift   |Parallel|
|N|Shift   |Perpendicular|
|T|Shift   |Tangent|
|M|Shift   |Midpoint|
|R|Shift   |Ratio|

mac4kent, 2022-08-09:
@amrsoll 
You still don't understand.
Deaf/Blind I have low vision and I unable to see too black edge. Please add feature so I able to change other color than black edge.
I am not talk about hotkey , shortcut. Only focus color to fix.

mac4kent, 2022-08-09:
Look at black edge I totally can't read. 
https://imgur.com/Y3cmy5z
See picture. 3D Viewer

amrsoll, 2022-08-09:
Ah, I see, my bad, I am looking into it

mac4kent, 2022-08-09:
Good finally you understand.

amrsoll, 2022-08-09:
Well, the color you are looking for is **Highlight**

[image]

[image]

Also I think the original bug has been fixed :) So we can close this issue?

amrsoll, 2022-08-09:
> Look at black edge I totally can't read.
https://imgur.com/Y3cmy5z

Also in your image, there is nothing inside the window, which explains why you cannot see anything :) You need to add something, which is why I gave you the shortcuts to add elements

mac4kent, 2022-08-09:
You already miss whole preferences I am show you again. 
[image]

I already setting there and still black after click plane I unable to see.

mac4kent, 2022-08-09:
Let me update again on CAD there update 2 hours ago.

amrsoll, 2022-08-09:
I don't see any black edges anywhere ... Could you draw a red circle around what you have problem seeing exactly?

amrsoll, 2022-08-09:
do not worry about the new updates, they are not related to the colors

mac4kent, 2022-08-09:
[image]

I see they add feature "Inactive" so I change from black to purple.

mac4kent, 2022-08-09:
Default I just change black to green.
There same I will check later after render is finish.

mac4kent, 2022-08-09:
I will wait for them to fix color full color without miss.

amrsoll, 2022-08-09:
> Look at black edge I totally can't read. 
https://imgur.com/Y3cmy5z See picture. 
3D Viewer

Please **draw red arrow** or **draw red circle around the black edge** and send it here.

I do not understand what it is you are trying to show me.

mac4kent, 2022-08-09:
Ok after render is finish then I will show you on youtube. I can do screen record on my iMac.

amrsoll, 2022-08-09:
From you previous video, here https://youtu.be/zCarMBGwUwE?t=300

You are circling the mouse around _nothing_. you just need to add new elements to the sketch you created :)

Example, try to press <kbd>L</kbd> and click somewhere on the screen :wink:

mac4kent, 2022-08-09:
Did you mean Push the Shift + Command + 5 then push the L key? (Mac)

amrsoll, 2022-08-09:
You can also find the buttons to add elements if you press the arrow on the left of your 3D view

[image]

You will see buttons like these ones for CAD Sketcher

[image]

amrsoll, 2022-08-09:
> Did you mean Push the Shift + Command + 5 then push the L key? (Mac)

Nope, just <kbd>L</kbd> then use your mouse to click in the 3D view :)

mac4kent, 2022-08-09:
How did you get red arrow?
No not talk about tools (you show me picture on left side). Focus on plane (floor) color.
Let me open Affinity Designer.app to show you.
1. There 3 plane and mouse over on plane become gray that one and I will click.
https://imgur.com/EwJcQjf

2. After click and I unable to see too black edge line of plane
https://imgur.com/HzGcCOX

amrsoll, 2022-08-09:
The plane disappeared. It is normal. Now you are inside the sketch

[image]

[image]

Now you need to add stuff to your sketch. 

I can recommend to follow the [tutorial](https://hlorus.github.io/CAD_Sketcher/getting_started/#create-a-sketch) 

> How did you get red arrow?

My screen capture program has that feature

amrsoll, 2022-08-09:
I am going to close this issue now, as the discussion is completely off the original topic.

amrsoll, 2022-08-09:
If you need help in general, you might have more luck on the official chat of CAD Sketcher :) 

https://discord.com/channels/882152824249667594/891060540498124800

closed 2022-08-09 by amrsoll

mac4kent, 2022-08-09:
Did you mean "My screen capture program has that feature"
look at this Keyboard shortcut 
[image]

amrsoll, 2022-08-09:
We do not have the save system. I don't use Mac.

mac4kent, 2022-08-09:
Sigh

mac4kent, 2022-08-09:
@amrsoll 
We know that Mac an Linux (There is NOT TRUE!!!!!!!!)
Mac and Linux is bigger different.
Linux is Windows.
Mac is Macintosh.
You know about Mac then you must have Mac computer like iMac, PowerBook Pro is Mac, MacPro is Mac.

mac4kent, 2022-08-09:
C:>/ is Dos and Windows and Linux.
Volumes is Mac.

mac4kent, 2022-08-09:
kentdavis@Kents-iMac / % cd /Volumes; ls
Kent Davis
kentdavis@Kents-iMac /Volumes %

amrsoll, 2022-08-09:
No Kent,
That is not what I wrote.
You are wrong about how Linux, Mac and Windows work.

I do not think you are doing it on purpose, but you are being difficult to help.

I am not going to answer further.

Good luck with using CAD sketcher for your project.

amrsoll, 2022-08-09:
@mac4kent 

> We know that Mac an Linux (There is NOT TRUE!!!!!!!!)
.
Actually you seem to be right. I made a typo.

Mac *and* Linux are very different.