hlorus/CAD_Sketcher #217: Can't edit or move new line dimension constraint

jbhome, 2022-07-05:
I was following the Maker Tales intro to Blender/CS, running latest CS.
I created rectangle and located it with the construction line. Then I tried to dimension one side. I picked the ends of the line and selected "Distance constraint" and it worked as expected.
I then tried to edit the distance value but it was not accessible.
I tried to move it's location and that also failed to work.
To compare, I repeated the steps to this stage and noticed, when working properly, the dimension line changed colour when the mouse hovered over it. This did not happen during the fault.
I have gone through the example several times successfully as I familiarize myself with the workflow but this time the distance constraint didn't work properly.
Closing and restarting Blender/CS fixed the issue, but I did not save the sketch before restarting so it was not really a fair test.
Is there some way of setting an activity log we can send to you to help in identifying and fixing issues?
Regards,
Jim Brooks

exegetor, 2022-07-11:
jbhome, I've seen this myself in recent days.   While we try to get it fixed, please be aware that ctrl-z (maybe several times)  will restore the state of the sketch to where they go live again.  Irritating, obviously, but it will save you from having to throw away work.
As hlorus said, we could really use logs and any ideas you have about what may be triggering this.

hlorus, 2022-07-11:
Btw which blender version are you using? This could be a duplicate of #103, make sure you're running a stable version of blender.

TrabacchinLuigi, 2022-07-27:
Hello, i think i'm in same situation i get this log:

```
Python: Traceback (most recent call last):
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 469, in modal
    return self.evaluate_state(context, event, event_triggered)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 551, in evaluate_state
    return self._end(context, True)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 612, in _end
    self.fini(context, succeede)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\operators\add_line_2d.py", line 83, in fini
    solve_system(context, sketch=self.sketch)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 271, in solve_system
    return solver.solve()
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 198, in solve
    self._init_slvs_data()
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 127, in _init_slvs_data
    indices = c.create_slvs_data(self.solvesys, group=group)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\class_defines.py", line 2200, in create_slvs_data
    return make_coincident(
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\class_defines.py", line 2154, in make_coincident
    handle = e2.py_data
AttributeError: 'NoneType' object has no attribute 'py_data'
```
that is the first error i get then i keep getting others that appear to have different stacks, i can provide the file, since i0ve saved it it won't let me sketch anymore

TrabacchinLuigi, 2022-07-27:
the file is here https://we.tl/t-5iyBrlRZnA

hlorus, 2022-08-04:
> Hello, i think i'm in same situation i get this log:
> 
> ```
> Python: Traceback (most recent call last):
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 469, in modal
>     return self.evaluate_state(context, event, event_triggered)
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 551, in evaluate_state
>     return self._end(context, True)
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 612, in _end
>     self.fini(context, succeede)
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\operators\add_line_2d.py", line 83, in fini
>     solve_system(context, sketch=self.sketch)
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 271, in solve_system
>     return solver.solve()
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 198, in solve
>     self._init_slvs_data()
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 127, in _init_slvs_data
>     indices = c.create_slvs_data(self.solvesys, group=group)
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\class_defines.py", line 2200, in create_slvs_data
>     return make_coincident(
>   File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\class_defines.py", line 2154, in make_coincident
>     handle = e2.py_data
> AttributeError: 'NoneType' object has no attribute 'py_data'
> ```
> that is the first error i get then i keep getting others that appear to have different stacks, i can provide the file, since i've saved it it won't let me sketch anymore
> 
> 

That doesn't seem to be related to the issue here, open another issue please.

StephenBrough, 2022-08-11:
I think I've encountered the same issue. I encountered the issue when I was working through a Maker Tales video and got to [this](https://youtu.be/1jNDLUDL0gc?list=PL6Fiih6ItYsXlk87595sDny7VnvQu3PoS&t=482) point. I added the distance dimension, the sketch became fully defined, and I couldn't edit or even select any of the constraints after that. Saving then restarting Blender and opening the save file did not fix the issue. 

 The `CAD_Sketcher-main.log` file in AppData/Local/Temp is empty. 

I'm running Blender v3.2.2 release (master).