hlorus/CAD_Sketcher #218: Code Cleanup

hlorus, 2022-07-07:
As the addon has grown quite a bit and new contributors are jumping onboard there's some cleanup overdue. This task is used to keep track of code cleanup efforts.

Feel free to suggest additional topics/areas in the comments.

## Structure
- [ ] Split out Stateful Op
- [ ] Improve registration flow (Static or auto register a module?  https://blender.stackexchange.com/questions/44532/best-practice-for-add-on-generation-with-many-functions-single-file-or-split-it)
- [ ] Slip out utilities and constants
- [ ] Split out blender classes into into individual files (PropertyGroups, Operators, Panels, etc...)

## Style
- [ ] Setup formatting (Guidline? Add a github workflow?)
- [ ] Add docstrings
- [ ] Add type hints

bonjorno7, 2022-08-10:
It helps a lot with maintaining readability if everyone formats their code the same way.
I use YAPF for formatting, which can be setup in most popular text editors.
Here are my VS Code settings for reference:
```json
{
    "editor.rulers": [
        80,
        120
    ],
    "editor.formatOnPaste": true,
    "editor.formatOnSave": true,
    "python.formatting.provider": "yapf",
    "python.formatting.yapfArgs": [
        "--style",
        "{based_on_style: google, column_limit: 120}"
    ],
    "python.sortImports.args": [
        "-l",
        "120"
    ],
    "[python]": {
        "files.insertFinalNewline": true,
        "files.trimFinalNewlines": true
    },
    "python.linting.enabled": true,
    "python.linting.pylintEnabled": false,
    "files.exclude": {
        "**/*.pyc": true
    }
}
```

hlorus, 2022-08-11:
> It helps a lot with maintaining readability if everyone formats their code the same way.
> I use YAPF for formatting, which can be setup in most popular text editors.
> Here are my VS Code settings for [...]

I agree, there should be a formatting rule. I would try to keep it as simple as possible tho so new contributors can easily jump on board without having to deal with alot of configuration first. IMO autopep8 would be a good choice as it's already the default setting in vscode. An alternative could be black where you don't have any configuration at all.

laundmo, 2022-08-11:
Another topic to consider is structuring the repository differently, if blender allows. Usually for normal python packages the top level should not include any python source files, instead they should be a src or cad_sketcher directory which contains all source files and a __main__.py entrypoint.

Along with this i would discourage the use of relative imports, as they can lead to issues with modules on sys.path having the same name and are discouraged by PEP8.

This all hinges on how blender works with addons, a topic which i have no experience with.

hlorus, 2022-08-11:
> Another topic to consider is structuring the repository differently, if blender allows. Usually for normal python packages the top level should not include any python source files, instead they should be a src or cad_sketcher directory which contains all source files and a __main__.py entrypoint.

Yeah i'm aware of that and i agree it would be nice to move files away from the root folder. The reason they live in the root folder is to support the git installation workflow mentioned [here](https://hlorus.github.io/CAD_Sketcher/installation/) where the addon is placed in the scripts folder defined in blenders preferences. I'va already tried to simply add a link on the root level to the __init__.py which blender looks for but that didn't seem to work unfortunately.

> Along with this i would discourage the use of relative imports, as they can lead to issues with modules on sys.path having the same name and are discouraged by PEP8.

True, i don't really see a reason against it 

> This all hinges on how blender works with addons, a topic which i have no experience with.

bonjorno7, 2022-08-11:
> Another topic to consider is structuring the repository differently, if blender allows. Usually for normal python packages the top level should not include any python source files, instead they [...]

For addon development, I think most people use Jacque Lucke's VS Code extension, which allows for convenient debugging and reloading.
It requires that the folder you're editing has an init.py with bl_info.
As for imports, Blender requires them to be relative; we can't follow pep8 in this regard.
If it works as an addon, it won't work standalone, and visa versa.

laundmo, 2022-08-11:
>  Blender addons only work with relative imports; we can't follow pep8 in this regard.

im curious about this: how does blender load addons? how can blender affect the choice of importing used?

bonjorno7, 2022-08-11:
> >  Blender addons only work with relative imports; we can't follow pep8 in this regard.
> 
> im curious about this: how does blender load addons? how can blender affect the choice of importing used?

I guess the addon's package isn't the top package, but one that is imported by Blender from higher up.
Therefore its submodules don't exist at the top level, but rather as `addon_name.sub_module` in `sys.modules`.

Cheaterman, 2023-02-14:
Let's clarify status on this :-) - do you consider #229 to be finished? Is there a WIP PR to split out utils/constants or is this a potential point where I could get started?

I'm not entirely sure I'll be able to singlehandedly add all necessary type hints and docstrings but a best effort will be made :-) (I think my time is best spent on implementing new features in the first place, then I'll use remaining available time to work on docs & types)

hlorus, 2023-02-14:
> Let's clarify status on this :-) - do you consider #229 to be finished? 

Yes that's already done.

> Is there a WIP PR to split out utils/constants or is this a potential point where I could get started?

This is also already done, mostly in this commit:
https://github.com/hlorus/CAD_Sketcher/commit/b73d41b1ce5749abb988533e4bb6f215e468353e#diff-8656a273c89d5812379185ade358273fa3d93e2cb8df832c66cec928eef9809a
 
> I'm not entirely sure I'll be able to singlehandedly add all necessary type hints and docstrings but a best effort will be made :-) (I think my time is best spent on implementing new features in the first place, then I'll use remaining available time to work on docs & types)

IMO this is just an ongoing topic that can be improved when doing cleanups or refactors.

What's mainly remaining here is to split ui.py and workspacetools.py into individual files and maybe move all keymap definitions to one place e.g. keymaps.py.

hlorus, 2023-02-14:
Also if you encounter stuff that could use a cleanup/refactor, here would be a good point to report that

PR #336 (2023-02-16): Split gizmos.py into submodules, final task of #218
Closes #218 when merged.
closed 2023-02-16 by Cheaterman