hlorus/CAD_Sketcher #224: [BUG] register_class already registered

ralfbiedert, 2022-07-17:
### Description

Enabling CAD Sketcher gives an exception:

[image]

### Addon Version

0.24.0

### Blender Version

3.2.0

### What platform are you running on?

Windows

hlorus, 2022-07-17:
Thanks for reporting,
Does this also happen when you restart blender after the installation? You might also try the latest version from github.

ralfbiedert, 2022-07-18:
> Does this also happen when you restart blender after the installation? 

Hm, yes, I've just opened up Blender again and now it seems to work. Trying the latest `main` gives this though:

[image]

hlorus, 2022-07-18:
Ahh i see, should be fixed by: https://github.com/hlorus/CAD_Sketcher/commit/2d6f60992087e0507a01d5b03761a12a2439c42b

closed 2022-07-18 by hlorus